raspberrypi/pico-sdk #700: fatal error: stdlib: No such file or directory 10 | #include <cstdlib>

rmacrino, 2022-01-04:
_common/pico_standard_link/new delete.c++:10:10: fatal error: stdlib: No such file or directory 10 | #include <cstdlib>
in wsl and in another machine with last version linux kali, same error the fils cstdlib found ,and in PATH....
I compile with verbose=1 and get the same errror

"......    -DLIB_PICO_STDLIB=1 -DLIB_PICO_SYNC=1 -DLIB_PICO_SYNC_CORE=1 -DLIB_PICO_SYNC_CRITICAL_SECTION=1 -DLIB_PICO_SYNC_MUTEX=1 -DLIB_PICO_SYNC_SEM=1 -DLIB_PICO_TIME=1 -DLIB_PICO_UTIL=1 -DPICO_BOARD=\"pico\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Release\" -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_PROGRAM_URL=\"https://github.com/raspberrypi/pico-examples/tree/HEAD/blink\" -DPICO_TARGET_NAME=\"blink\" -DPICO_USE_BLOCKED_RAM=0 -I/pico/pico-sdk/src/common/pico_stdlib/include -I/pico/pico-sdk/src/rp2_common/hardware_gpio/include -I/pico/pico-sdk/src/common/pico_base/include -I/pico/pico-examples/build/generated/pico_base -I/pico/pico-sdk/src/boards/include -I/pico/pico-sdk/src/rp2_common/pico_platform/include -I/pico/pico-sdk/src/rp2040/hardware_regs/include -I/pico/pico-sdk/src/rp2_common/hardware_base/include -I/pico/pico-sdk/src/rp2040/hardware_structs/include -I/pico/pico-sdk/src/rp2_common/hardware_claim/include -I/pico/pico-sdk/src/rp2_common/hardware_sync/include -I/pico/pico-sdk/src/rp2_common/hardware_uart/include -I/pico/pico-sdk/src/rp2_common/hardware_divider/include -I/pico/pico-sdk/src/common/pico_time/include -I/pico/pico-sdk/src/rp2_common/hardware_timer/include -I/pico/pico-sdk/src/common/pico_sync/include -I/pico/pico-sdk/src/common/pico_util/include -I/pico/pico-sdk/src/rp2_common/pico_runtime/include -I/pico/pico-sdk/src/rp2_common/hardware_clocks/include -I/pico/pico-sdk/src/rp2_common/hardware_irq/include -I/pico/pico-sdk/src/rp2_common/hardware_resets/include -I/pico/pico-sdk/src/rp2_common/hardware_pll/include -I/pico/pico-sdk/src/rp2_common/hardware_vreg/include -I/pico/pico-sdk/src/rp2_common/hardware_watchdog/include -I/pico/pico-sdk/src/rp2_common/hardware_xosc/include -I/pico/pico-sdk/src/rp2_common/pico_printf/include -I/pico/pico-sdk/src/rp2_common/pico_bootrom/include -I/pico/pico-sdk/src/common/pico_bit_ops/include -I/pico/pico-sdk/src/common/pico_divider/include -I/pico/pico-sdk/src/rp2_common/pico_double/include -I/pico/pico-sdk/src/rp2_common/pico_int64_ops/include -I/pico/pico-sdk/src/rp2_common/pico_float/include -I/pico/pico-sdk/src/rp2_common/pico_malloc/include -I/pico/pico-sdk/src/rp2_common/boot_stage2/include -I/pico/pico-sdk/src/common/pico_binary_info/include -I/pico/pico-sdk/src/rp2_common/pico_stdio/include -I/pico/pico-sdk/src/rp2_common/pico_stdio_uart/include -mcpu=cortex-m0plus -mthumb -O3 -DNDEBUG -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-rtti -fno-use-cxa-atexit -std=gnu++17 -MD -MT blink/CMakeFiles/blink.dir/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj -MF CMakeFiles/blink.dir/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj.d -o CMakeFiles/blink.dir/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj -c /pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp
/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:10:10: fatal error: cstdlib: No such file or directory
   10 | #include <cstdlib>
      |          ^~~~~~~~~
compilation terminated.
make[2]: *** [blink/CMakeFiles/blink.dir/build.make:653: blink/CMakeFiles/blink.dir/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj] Error 1
make[2]: se sale del directorio '/pico/pico-examples/build'
make[1]: *** [CMakeFiles/Makefile2:4119: blink/CMakeFiles/blink.dir/all] Error 2
make[1]: se sale del directorio '/pico/p"

kilograham, 2022-01-04:
i presume you do not have a arm-eabi-none-g++ or (libstd++) installed properly.

see also #387

lurch, 2022-01-04:
>  linux kali

See #645

rmacrino, 2022-01-05:
dpkg -L libstdc++-arm-none-eabi-newlib | grep cstdlib
/usr/include/newlib/c++/8.3.1/cstdlib
/usr/include/newlib/c++/8.3.1/tr1/cstdlib
are installed properly
but I am thinking that you are looking for a newer version, if so, it is a Kali bug

aallan, 2022-01-05:
This is an OS not SDK problem.

closed 2022-01-05 by aallan