raspberrypi/pico-sdk #703: Pico support in Embeetle IDE: how to view compiler commands?

kristofmulier, 2022-01-10:
<h2>1. Context</h2>

I'm cofounder of this new free microcontroller IDE:
https://embeetle.com

I'm adding support for the Raspberry Pi Pico. All projects in Embeetle are makefile-based. Therefore, I need to convert the Pi Pico sample projects from CMake-based to makefile-based.

<sup>PS: I know that CMake does generate makefiles, so the idea that I want to convert a CMake-based project into a makefile-based project might seem odd at first. The point is: I want to convert it into a project with a single, central human-readable makefile that invokes all the compiler calls.</sup>

<h2>2. Question</h2>

I found the compiler flags for the blinky sample project in the file `~/pico/pico-examples/build/blink/CMakeFiles/blink.dir`. That helps me already a great deal. However, I would like to see the exact compiler commands that the CMake build system invokes. However, when I do a build, I only see this:

```bash
$ cd ~/pico/pico-examples/build/blink
$ make
Creating directories for 'ELF2UF2Build'
No download step for 'ELF2UF2Build'
No update step for 'ELF2UF2Build'
No patch step for 'ELF2UF2Build'
Performing configure step for 'ELF2UF2Build'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kristof/pico/pico/pico-examples/build/elf2uf2
Performing build step for 'ELF2UF2Build'
[100%] Built target elf2uf2
No install step for 'ELF2UF2Build'
Completed 'ELF2UF2Build'
Built target ELF2UF2Build
Built target bs2_default
Built target bs2_default_padded_checksummed_asm
Building C object blink/CMakeFiles/blink.dir/blink.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_claim/claim.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_platform/platform.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_sync/sync.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_uart/uart.c.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_divider/divider.S.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_time/time.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_time/timeout_helper.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_timer/timer.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_sync/sem.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_sync/lock_core.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_sync/mutex.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_sync/critical_section.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_util/datetime.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_util/pheap.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/common/pico_util/queue.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_clocks/clocks.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_irq/irq_handler_chain.S.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_pll/pll.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_vreg/vreg.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_watchdog/watchdog.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/hardware_xosc/xosc.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_printf/printf.c.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_bit_ops/bit_ops_aeabi.S.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_bootrom/bootrom.c.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_divider/divider.S.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_double/double_aeabi.S.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_double/double_init_rom.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_double/double_math.c.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_double/double_v1_rom_shim.S.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_int64_ops/pico_int64_ops_aeabi.S.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_float/float_aeabi.S.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_float/float_init_rom.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_float/float_math.c.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_float/float_v1_rom_shim.S.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_mem_ops/mem_ops_aeabi.S.obj
Building ASM object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_standard_link/crt0.S.obj
Building CXX object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_standard_link/binary_info.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_stdio/stdio.c.obj
Building C object blink/CMakeFiles/blink.dir/home/kristof/pico/pico/pico-sdk/src/rp2_common/pico_stdio_uart/stdio_uart.c.obj
Linking CXX executable blink.elf
Built target blink
```

In other words - the build system tells me which files are being built, but not what the exact compiler calls are. I looked for a verbosity switch, but without success.

How can I instruct the build system to print all commands?

kristofmulier, 2022-01-10:
I found the solution. I just need to invoke the build with:

```bash
$ make VERBOSE=1;
```

Apologies.

closed 2022-01-10 by kristofmulier

kilograham, 2022-01-10:
Yes, using makefiles directly is strongly discouraged as it is rather brittle (to do it correctly youd have to rewrite all the cmake logic in make). The compile flags are dependent on what combination of libraries you are building, and of course any user specified flags. They may also change from one SDK release to another.

Depending on what you are doing, you may be able to come up with a common set of flags

kristofmulier, 2022-01-12:
Hi @kilograham ,
It is not my intention to reproduce the whole CMake logic in a makefile. We want to create a simple makefile-based blinky project that *just works* on a Pi Pico. Now that I can see the verbose output of the build, I'm making progress to get there. 

Meanwhile, I noticed that the CMake build system is quite complex. It seems to generate a couple of source files on-the-fly. For example: it generates this `../pico-sdk/src/rp2_common/boot_stage2/bs2_default_padded_checksummed.S` file that is then given to the linker command.

Any pitfalls I should be aware of in my attempt to create a makefile-based blinky project that *just works*?