raspberrypi/pico-sdk #704: Pico support in Embeetle IDE: Is the file 'bs2_default_padded_checksummed.S' really needed?

kristofmulier, 2022-01-12:
<h2>1. Context</h2>

I'm cofounder of this new free microcontroller IDE:
https://embeetle.com

I'm adding support for the Raspberry Pi Pico. All projects in Embeetle are makefile-based. Therefore, I need to convert the Pi Pico sample projects from CMake-based to makefile-based.

It is not my intention to rewrite the CMake logic in a makefile. I just want to create a simple makefile-based blinky LED project that *just works*.

<sup>PS: I know that CMake does generate makefiles, so the idea that I want to convert a CMake-based project into a makefile-based project might seem odd at first. The point is: I want to convert it into a project with a single, central human-readable makefile that invokes all the compiler calls.</sup>

<h2>2. Situation so far</h2>

I followed this tutorial:
https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

I now have the folders `~/pico/pico-sdk` and `~/pico/pico-examples`. To create all the makefiles - in other words: to launch cmake - I do this:

```bash
$ cd pico-examples
$ mkdir build
$ cd build
$ cmake ..
```

To build the blinky sample project, I navigate into the folder `~/pico/pico-examples/build/blink` and issue the command:

```bash
$ make VERBOSE=1;
```

It's this blinky sample project that I want to convert into a simple makefile-based project for our Embeetle IDE users.

<h2>3. About 'bs2_default_padded_checksummed.S'</h2>

In the verbose build output, I can see that the file `bs2_default_padded_checksummed.S` is passed to the linker at the very end. I figured out that this file is generated by a python script at `~pico/pico-sdk/src/rp2_common/boot_stage2/pad_checksum`. I found this recipe in `~/pico/pico-examples/build/pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default_padded_checksummed_asm.dir/build.make`:

```makefile
pico-sdk/src/rp2_common/boot_stage2/bs2_default_padded_checksummed.S: pico-sdk/src/rp2_common/boot_stage2/bs2_default.bin
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --blue --bold --progress-dir=/home/kristof/pico/pico-examples/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_1) "Generating bs2_default_padded_checksummed.S"
	cd /home/kristof/pico/pico-examples/build/pico-sdk/src/rp2_common/boot_stage2 && /usr/bin/python3.9 /home/kristof/pico/pico-sdk/src/rp2_common/boot_stage2/pad_checksum -s 0xffffffff /home/kristof/pico/pico-examples/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default.bin /home/kristof/pico/pico-examples/build/pico-sdk/src/rp2_common/boot_stage2/bs2_default_padded_checksummed.S
```

<h2>4. Question</h2>

We don't want to make any assumption about the user's local installation. For example, Embeetle IDE doesn't require Python to be installed. Generating the `bs2_default_padded_checksummed.S` file with the Python script is therefore not an option. So I wonder:

 - Is this file `bs2_default_padded_checksummed.S` absolutely necessary? What is it actually for?

 - Can I generate the file beforehand and put it in the user's project? Probably the checksum is computed based on the user's code and therefore it's impossible to create the file beforehand.

Thank you very much for your help.

lurch, 2022-01-12:
> Is this file bs2_default_padded_checksummed.S absolutely necessary? What is it actually for?

Have a look at "2.8.1. Processor Controlled Boot Sequence" in https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf

> Can I generate the file beforehand and put it in the user's project? Probably the checksum is computed based on the user's code and therefore it's impossible to create the file beforehand.

That's more a question for @kilograham , but if you're only supporting a single boot-stage-2 (i.e. you don't allow your users to use an RP2040 board with a different flash chip than the one used on the Raspberry Pi Pico boards), then yes you _might_ be able to ship a pre-checksummed boot-stage-2. Maybe :wink:

JamesH65, 2022-01-12:
There is no flash on the RP2040 itself, so yes, the flash chip will have to be used.

You could load directly to RAM using SWD/Debugger if you want.

lurch, 2022-01-12:
> I didn't notice this beforehand, but the Pico Pi board has indeed a separate Flash chip: W25Q16JVUXIQ. If you flash one of the sample projects (eg. the blinky one in the SDK), will this Flash chip be used?

Yes, https://github.com/raspberrypi/pico-sdk/blob/master/src/board_setup.cmake#L7 sets the default PICO_BOARD to `pico`, and https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/pico.h#L67 then configures which boot-stage-2 to use.

> My goal is to create a makefile-based sample project that just works on the Pico Pi board. We can add a comment to the boot file, to warn the advanced users who want to connect the RP2040 chip to another separate Flash chip.

Hopefully you're already aware, but there are a number of 3rd-party boards using RP2040, each of which may be using a different flash chip, so this isn't necessarily an "advanced users" thing. See https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards and the `PICO_BOOT_STAGE2_CHOOSE_` used by each.

kristofmulier, 2022-01-12:
> *"There is no flash on the RP2040 itself, so yes, the flash chip will have to be used."*

I didn't know that (I'm new to the RP2040). I'm used to chips like the STM32 series, that have on-chip flash memory.

> *"Hopefully you're already aware, but there are a number of 3rd-party boards using RP2040, each of which may be using a different flash chip, so this isn't necessarily an "advanced users" thing."*

Our users can select a board and chip in the IDE. We could create sample projects for a few of those 3rd-party boards too, and provide the checksummed 2nd stage bootloader file for each.