raspberrypi/pico-sdk #710: USB host and Hub support

jgilje, 2022-01-16:
Just wondered if anyone's got host support working with a HID device connected behind a Hub. I'm working from the host_cdc_msc_hid example. When testing, the device gets stuck in some sort of receive-timeout loop.

As far as I know the USB protocol, everything from TinyUSB is correct, but enumeration fails as soon as the hub port has been enabled. I'm running with CFG_TUSB_DEBUG set to 3. After the hub port has been enabled, the following log gets dumped to the serial debug line - after which further USB processing stops.
```
Open EP0 with Size = 8 (addr = 0)
hcd_edpt_open dev_addr 0, ep_addr 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Get 8 byte of Device Descriptor
Control Setup (addr = 0): 80 06 00 01 00 00 08 00 
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
```

At this point, `USB_INTS_ERROR_RX_TIMEOUT_BITS` is continuously signaled to the USB IRQ handler `hcd_rp2040_irq()`, in `hcd_rp2040.c` around line 223. I've tried with the TinyUSB stack as shipped in SDK release 1.3.0, and I've also tested using the latest master from TinyUSB without any changes in behaviour.

lurch, 2022-01-16:
ping @hathach

kilograham, 2022-01-17:
cc @liamfraser

hathach, 2022-01-17:
will check this out whenever I could, being busy with other works at the moment.

liamfraser, 2022-01-17:
Can you please try out this branch @jgilje https://github.com/liamfraser/tinyusb/tree/rpikbd-rebase - I have several fixes there I am intending to PR into TinyUSB

jgilje, 2022-01-17:
@liamfraser just tested with your branch, I still have the same issue at my side.

jgilje, 2022-03-16:
@hathach Hi, just wanted to let you know after [1915d69cb8](https://github.com/hathach/tinyusb/commit/1915d69cb8da8372b35a5067420c57ddb65d2cb1) in your main repo, the issue mentioned here is solved.

hathach, 2022-03-17:
thank for the update, I totally forget about this :)

closed 2022-03-17 by jgilje