raspberrypi/pico-sdk #714: Can't install pico-sdk on Mac

mroberts1, 2022-01-21:
I'm unable to build the examples on a MacOS 12.1. 

I have cmake installed and clone the repo to my user account, but running cmake .. just results in cmake not being able to find either CMakeLists.txt, even after I move it to the ~/ directory, or the pico-sdk directory, wherever I put it.

I set the path for SDK in .zshrc but it still doesn't work.

Part of the problem seems to be that the downloaded repo isn't even called pico-sdk, but picosystem. The CMakeLists.txt file can't find the locations specified.

I currently have the picosystem repo in my root user directory. Do I need to rewrite CMakeLists.txt to get it to compile the examples? All the dependencies appear to be in place.

aallan, 2022-01-21:
> the downloaded repo isn't even called pico-sdk, but picosystem.

I'd guess you haven't downloaded the right repo at this point, this should not be the case.

lurch, 2022-01-21:
A quick google-search finds https://github.com/pimoroni/picosystem so if you're having trouble getting that set up, it sounds like you ought to be asking Pimoroni for help.

mroberts1, 2022-01-22:
> A quick google-search finds https://github.com/pimoroni/picosystem so if you're having trouble getting that set up, it sounds like you ought to be asking Pimoroni for help.

That's the device's operating system, whereas I'm trying to set up the sdk.

lurch, 2022-01-22:
> whereas I'm trying to set up the sdk.

In that case you probably want to read https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf (which is linked to from https://www.raspberrypi.com/documentation/microcontrollers/rp2040.html#documentation )

lurch, 2022-03-22:
Any more comments @mroberts1 or can this be closed now?

closed 2022-07-11 by kilograham