raspberrypi/pico-sdk #716: Documentation for gpio_set_irq_enabled doesn't specify the mask values, only bit number

danieljabailey, 2022-01-23:
The documenation for gpio_set_irq_enabled is a little bit unhelpful in that it doesn't speicfy the mask numbers (which are the numbers that someoe is interested in when they read the docs for this function), it only specifies the bit values.

It also fails to mention the enum gpio_set_irq_enabled which defines the mask values.

I have a PR ready that fixes this.

danieljabailey, 2022-01-23:
PR: #717

lurch, 2024-05-20:
I guess this ought to be closed, given that the linked PR #717 has been closed?

kilograham, 2026-09-30:
Fixed by #850 (in 1.4.0): the `gpio_set_irq_enabled` docs now reference `enum gpio_irq_level`, which lists the mask values. Please reopen if you think anything is still missing.

closed 2026-09-30 by kilograham