raspberrypi/pico-sdk #724: irq_add_shared_handler() causes assertion

baoshi, 2022-02-09:
Calling  irq_add_shared_handler() on same IRQ causes assertion error. It happens for both head and v1.3. Here is some sample code:

```
#include "pico/stdlib.h"
#include "hardware/dma.h"
#include "hardware/irq.h"

void dma_isr1()
{
    dma_hw->ints0 = 0; // actual code clear the bit only
}

void dma_isr2()
{
    dma_hw->ints0 = 0; // actual code clear the bit only
}

int main() 
{
    irq_add_shared_handler(DMA_IRQ_0, dma_isr1, PICO_SHARED_IRQ_HANDLER_DEFAULT_ORDER_PRIORITY);
    irq_add_shared_handler(DMA_IRQ_0, dma_isr2, PICO_SHARED_IRQ_HANDLER_DEFAULT_ORDER_PRIORITY);
    while(1) {}
}
```

Second call for irq_add_shared_handler causes assertion error at 
irq_add_shared_handler in irq.c line 240
make_branch() in irq.c line 139 (SDK 1.3)  
assert(!(delta >> 11u));

JamesH65, 2022-02-09:
You need to remove the handler for the specific IRQ before reassigning it. See irq_remove_handler(). Assigning two handlers to one IRQ is not supported.

kilograham, 2022-02-09:
is it fair to say that this actually works (in release mode), and that it is just the assertion that is wrong.

kilograham, 2022-05-04:
merged into develop

PR #747 (2022-04-29): irq_add_shared_handler assertion fix #724
fixes #724
commit 5a427fbf (2022-04-29): irq_add_shared_handler assertion fix #724 (#747)
closed 2022-05-04 by kilograham