raspberrypi/pico-sdk #725: CMake / NMake issue on windows 10

rwboyer, 2022-02-09:
Don't really know where to start with this issue but I've not seen it happen with Nmake before so I will start here with the SDK.

I have build environments setup correctly and working on Linux/OS X and even windows for the pico SDK but on windows I cannot get a pico build environment to work with Nmake with either the samples or anything else with the Pico SDK.

1. Using the sample code and SDK directly from the repository CMake -G "NMake Makefiles" works fine and builds all the make files.
2. Nmake works fine and builds the SDK object files but fails when attempting to link the object files failing to "find" the example code object file (which is present and built).

Nothing seems to be wrong with the build.make file in build/Cmakefiles/blink.dir generated from CMake that I can see:

`@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --bold --progress-dir=C:\Users\rwboy\pico\build-test\blink-ms\build\CMakeFiles --progress-num=$(CMAKE_PROGRESS_47) "Linking CXX executable blink.elf"`

Here's the build error, note the strange characters on line 2???:

`[build] [ 22%] Linking CXX executable blink.elf
[build] AR10B2~1.EXE: error: ∩╗┐CMakeFiles/blink.dir/blink.c.obj: No such file or directory
[build] NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102021~1.10\bin\AR10B2~1.EXE' : return code '0x1'
`
The RSP file containing all the object files to link appears to be fine as well.

I have a gaggle of linux/bsd/OS X dev/build environments and can also work around this on Windows 10 using gcc make instead of Microsoft Nmake but would love to run this down.

Any ideas why this is happening?

rwboyer, 2022-02-09:
@aallan 

Ps. works fine with GCC tools make on same machine.

kilograham, 2022-02-09:
CMake versions on windows after 3.19.x are broken.
Either downgrade, or you can use ninja which is much faster anyway.

aallan, 2022-02-09:
> CMake versions on windows after 3.19.x are broken. Either downgrade, or you can use ninja generator with the later cmake which is much faster anyway.

Oh that was it. I really need to add something to the Getting Started guide that mentions this problem.

kilograham, 2023-02-09:
think this is answered

closed 2023-02-09 by kilograham