raspberrypi/pico-sdk #729: add config option to make absolute_time_t always a uint64_t

kilograham, 2022-02-10:
The intention was that people would always use the macros (e.g. `to_us_since_boot`, so the difference in implementation would not matter. 

As stated in the docs the reason for making absolute_time_t a separate type was to avoid confusion between timeouts and longs.

Frankly, if people were using the macros, it is probably fine to make the default be that everything is a uint64_t.

noname22, 2023-11-11:
> As stated in the docs the reason for making absolute_time_t a separate type was to avoid confusion between timeouts and longs.

Where is this explained in the documentation? I can't find any explanation of the rational.

kilograham, 2024-07-07:
it is in the doxygen for the type

lurch, 2024-07-07:
...which can be seen online at https://www.raspberrypi.com/documentation/pico-sdk/group__timestamp.html#rpip6054cfaa87d222c22077

kilograham, 2024-07-07:
but yes we will make this off by default

closed 2024-08-08 by kilograham