raspberrypi/pico-sdk #734: Missing 'pwm_config_set_clkdiv_int_fract()' function?

ZodiusInfuser, 2022-02-21:
I've been dealing with the PWM hardware a lot in recent weeks for an upcoming Pimoroni product, and noticed that there seems to be no way set a clock divisor with a fractional integer argument during configuration.

There's a `pwm_config_set_clkdiv_int(pwm_config * c, uint div)` but no `int_fract` equivalent. Instead I am having to either cast my ints to floats to use `pwm_config_set_clkdiv(pwm_config * c, float div)`, or activate the PWM and change the frequency using `pwm_set_clkdiv_int_frac(uint slice_num, uint8_t integer, uint8_t fract)`.

Best as I can tell, there is no hardware limitation for this so it may just be an oversight. If this is the case would you like me to raise a PR to add it? I would leave the existing function alone for compatibility of course.

lurch, 2022-02-21:
In a similar vein, there's no `pwm_set_clkdiv_int` either :smiling_imp: 
So yeah, looks like a small oversight that the `pwm_config_set_` functions have `clkdiv` and `clkdiv_int`; but the `pwm_set_` functions have `clkdiv` and `clkdiv_int_frac` ? :shrug:

ZodiusInfuser, 2022-02-21:
You make a valid point re: missing `pwm_set_clkdiv_int()` 😆 .

I mention the lack of the `fract` version more so because hardware features like PIO have both a `sm_config_set_clkdiv_int_frac()` and `pio_sm_set_clkdiv_int_frac()`, leading me to believe that's the intended convention.

I don't think there's an `int` only version anywhere else in the API, but I'd be happy to be proven wrong 🙂

lurch, 2022-02-21:
Yup, looks like you're right:
```bash
$ git grep "_int(" | grep -v resolvable_int
src/common/pico_binary_info/include/pico/binary_info/code.h:#define bi_int(_tag, _id, _value) \
src/common/pico_binary_info/include/pico/binary_info/code.h:#define bi_binary_end(end) bi_int(BINARY_INFO_TAG_RASPBERRY_PI, BINARY_INFO_ID_RP_BINARY_END, end)
src/rp2_common/boot_stage2/pad_checksum:def any_int(x):
src/rp2_common/hardware_pwm/include/hardware/pwm.h:static inline void pwm_config_set_clkdiv_int(pwm_config *c, uint div) {
src/rp2_common/hardware_pwm/include/hardware/pwm.h:    pwm_config_set_clkdiv_int(&c, 1);
$ git grep "_int_frac("
src/rp2_common/hardware_pio/include/hardware/pio.h:static inline void sm_config_set_clkdiv_int_frac(pio_sm_config *c, uint16_t div_int, uint8_t div_frac) {
src/rp2_common/hardware_pio/include/hardware/pio.h:    sm_config_set_clkdiv_int_frac(c, div_int, div_frac);
src/rp2_common/hardware_pio/include/hardware/pio.h:    sm_config_set_clkdiv_int_frac(&c, 1, 0);
src/rp2_common/hardware_pio/include/hardware/pio.h:static inline void pio_sm_set_clkdiv_int_frac(PIO pio, uint sm, uint16_t div_int, uint8_t div_frac) {
src/rp2_common/hardware_pio/include/hardware/pio.h:    pio_sm_set_clkdiv_int_frac(pio, sm, div_int, div_frac);
src/rp2_common/hardware_pwm/include/hardware/pwm.h:static inline void pwm_set_clkdiv_int_frac(uint slice_num, uint8_t integer, uint8_t fract) {
src/rp2_common/hardware_pwm/include/hardware/pwm.h:    pwm_set_clkdiv_int_frac(slice_num, i, f);
test/hardware_pwm_test/hardware_pwm_test.c:            pwm_set_clkdiv_int_frac(pwm, i, f);
```

So _perhaps_ it makes sense to add `pwm_config_set_clkdiv_int_frac(pwm_config *c, uint8_t integer, uint8_t fract)`, and then make `pwm_config_set_clkdiv_int(pwm_config *c, uint div)` simply do `pwm_config_set_clkdiv_int_frac(c, div, 0);` ?

kilograham, 2022-02-21:
yes please add it via PR

kilograham, 2022-05-04:
merged into develop

closed 2022-05-04 by kilograham