raspberrypi/pico-sdk #738: Contradictory comments in boot2_w25q080.S and exit_from_boot2.S

Kristine1975, 2022-02-23:
According to the comment `boot2_w25q080.S`, upon entry `lr` is `0x10000100`:

https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2_common/boot_stage2/boot2_w25q080.S#L95-L96

According to the comment in `exit_from_boot2.S` however, it is zero, and that is what the code checks for:

https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2_common/boot_stage2/asminclude/boot2_helpers/exit_from_boot2.S#L12-L13

So my guess is the comment in `boot2_w25q080.S` is incorrect (as well as the one in `boot2_at25sf128a.S`).

lurch, 2022-02-24:
_Perhaps_ this is related to #10 ? :man_shrugging:

Kristine1975, 2022-02-24:
> _Perhaps_ this is related to #10 ? 🤷‍♂️

It doesn't look like it: The `0x10000100` comment already existed in the SDK's initial release 26653ea:

https://github.com/raspberrypi/pico-sdk/blob/26653ea81e340cacee55025d110c3e014a252a87/src/rp2_common/boot_stage2/boot2_w25q080.S#L94-L95

#10 merely extracted the code with the second comment to its own file `exit_from_boot2.S`.

closed 2024-08-08 by kilograham