raspberrypi/pico-sdk #742: tinyusb issue

Nedemai, 2022-02-27:
TinyUSB Host CDC MSC HID Example
assertion "ep->active" failed: file "C:\pico-sdk\lib\tinyusb\src\portable\raspberrypi\rp2040\hcd_rp2040.c", line 164, function: hw_trans_complete

I updated the SDK this morning and started getting this issue. Happens with any USB device I plug in, whether it be a mouse, keyboard or gamepad.

kilograham, 2022-02-27:
the bug is in TinyUSB... i think you can workaround it for now by switching to a release build (or removing the offending assert)

Nedemai, 2022-02-27:
As far as I understand I am using a release build.

kilograham, 2022-02-27:
well assert is a no-op in release builds (which define preprocessor variable `NDEBUG`)

kilograham, 2022-02-27:
If you aren't using an IDE, then make sure you pass `-DCMAKE_BUILD_TYPE=Release` to cmake

closed 2022-02-28 by Nedemai