raspberrypi/pico-sdk #750: cancel_repeating_timer not work well when the timer's delay is 1us

dty717, 2022-03-07:
When the timer's delay_us is 1, the callback's cancel_repeating_timer will not work.
```
bool repeating_accelerate_motor_callback(struct repeating_timer *t)
{
    motorToggleVal++;
    if (motorToggleVal < motorAccelerateInit)
    {
        gpio_set_dir(MOTOR_PWM_PIN, 0);
    }
    else if (motorToggleVal < 2 * motorAccelerateInit)
    {
        gpio_set_dir(MOTOR_PWM_PIN, 1);
    }
    else
    {
        motorToggleVal = 0;
        motorAccelerateCircles++;
        if (motorAccelerateCircles >= motorAccelerateStep)
        {
            printf("motorAccelerateInit:%d\r\n",motorAccelerateInit);
            motorAccelerateCircles = 0;
            if (motorAccelerateInit <= motorAccelerateFinal)
            {
                if (motorAccelerate)
                {
                    motorAccelerate = FALSE;
                    printf("motorAccelerateFinal:%d,motorMaxVal:%d\r\n", motorAccelerateFinal, motorMaxVal);
                    cancel_repeating_timer(&motorAccelerateTimer);
                    motorToggleVal = 0;
                    motorVal = 0;
                }else{
                    printf("error======> motorAccelerateFinal:%d,motorMaxVal:%d\r\n", motorAccelerateFinal, motorMaxVal);
                }
            }else{
                motorAccelerateInit--;
            }
        }
    }
    return true;
}
```
Adding the timer,
`
add_repeating_timer_us(1, repeating_accelerate_motor_callback, NULL, &motorAccelerateTimer);
`

Reneg973, 2022-03-07:
Of course. I assume it works better if you remove printf, which is useless in a 1e-6s callback.

dty717, 2022-03-07:
@Reneg973 you're right. I fail to express the question rightly.

lurch, 2022-03-07:
Vaguely similar to #737 ?

kilograham, 2023-02-09:
closing for now as returning false is the idiomatic way to cancel the timer

closed 2023-02-09 by kilograham