raspberrypi/pico-sdk #751: stdio_usb, hitting a 256 byte limit

shabaz123, 2022-03-08:
Hello,

I have some code which uses multiple **print** statements within a timer callback (using the **add_alarm_in_ms** function).
Each printf statement is not particularly long, but I have multiple printf because I'm printing a user menu. In total, across all the printf statements, the number of characters exceeds 256 bytes.

What occurs, is that only 256 bytes are printed. I have tried interspersing **sleep_ms** commands, but it does not like that (guessing that I cannot use the sleep_ms within a callback).

Anyway, I have a sort of workaround, which is to increase the 256 byte limit in the file 
**rp2_common/pico_stdio_usb/include/tusb_config.h**
so that a line in that file becomes this:

**#define CFG_TUD_CDC_TX_BUFSIZE  (1024)**

Is there a bug somewhere? Or am I doing something wrong? What's the proper way I should resolve this? I'd like to continue using the printf commands in the callback, because it suits the structure of the particular code I'm using. Agree it's not usually good practice to do a lot in an interrupt-like callback, but I don't re-enable the timer except at the end of the callback using the return value, i.e.:
**return(MY_ALARM_USEC_PERIOD);**

Thanks for your help.

lurch, 2022-03-08:
ping @hathach

kilograham, 2022-03-08:
> ping @hathach

not related to TinyUSB particularly

kilograham, 2022-03-08:
As you have guess your problem is caused by doing `printf` to USB during an IRQ callback. Your IRQ is higher priority than that the handles USB interrupts (and to a lesser extend that does TinyUSB backround work for `stdio_usb`).. you would generally deadlock except that we have deliberately tried to avoid toomuch foot-gunning in that arena, by making s stuff discard output instead.

The simplest way to workaround your problem is to create a new alarm pool via `alarm_pool_create()` using a different hardware alarm and then lower the priority of the IRQ for that alarm via `irq_set_priority(TIMER_IRQ_<N>, 0xC0)` or something and create your alarm callback using this new lower priority alarm pool

kilograham, 2022-03-08:
closing as it isn't really an issue per se.

closed 2022-03-08 by kilograham

shabaz123, 2022-03-08:
Hi @kilograham and @lurch , thanks for the response!
That makes sense, I've looked at the API documentation to understand the functions you mention, and I think I understand how to go about this now. I would very much like the priority of the alarm that I'm using to be lower than USB interrupt requests for this particular use-case, so what you say sounds like a good plan.

shabaz123, 2022-03-08:
Just in case anyone else hits this problem, the procedure described above worked : )
This was the functioning code:

    **alarm_pool = alarm_pool_create(2, 16); // create an alarm pool, use hardware alarm #2
    irq_set_priority(TIMER_IRQ_2, 0xc0); // larger number is lower priority
    alarm_pool_add_alarm_in_ms(alarm_pool, ALARM_MSEC_PERIOD, your_callback_function, NULL, false); // ALARM_MSEC_PERIOD can be say 20 for 20 msec**
    
And then in the callback which has function prototype:
**int64_t your_callback_function(alarm_id_t id, void *user_data)**

at the end simply **return (ALARM_MSEC_PERIOD)** so that the alarm is re-armed.