raspberrypi/pico-sdk #752: GPIO open drain mode

ulri-me, 2022-03-08:
Hi RP2040 devs,

I carefully read the RP2040 documentation about hardware_gpio and PIO, but I never read any word about Open Drain output mode for GPIO modes. Is this possible?

In my case I need to control a LED with open-drain mode.

It's definitely possible as there is e.g. I2C available, but I do not know how to access it.

Please kindly help me!

Kind regards
Robin

ulri-me, 2022-03-08:
Just to inform other users:
Every Push-Pull output as the RP2040 has, is also a Open-Drain-Output.

Just invert the signal to what value you would use if it was a push-pull-output.

The image of the second post at
https://circuitdigest.com/forums/embedded/difference-between-open-drain-and-push-pull#forum-topic-top
explains it.

closed 2022-03-08 by ulri-me

romsom, 2022-09-05:
I was looking for that, too. But as neither API docs nor [datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf) for the RP2040 mention open-drain operation, I assume it's just not implemented in the hardware design of the software controlled GPIO controller and only active when the I²C function is selected for the pin.

How does inverting the signal make a push-pull an open-drain output? There is an image, but it only shows the basic schematic for the two kinds of output stage.

You can add a transistor to the GPIO and have that as an open-drain/open-collector stage, though. Because that's also an inverter, you have to invert the signal in software as well.

romsom, 2022-10-25:
Oops, I completely forgot to test this. Thanks for reminding me! ;)
Maybe I'll have the time to test on Friday. If you're impatient and/or daring you can try it yourself.
This is completely untested(!), but I'd try this in C:

```
void set_gpio_open_drain(uint gpio) {
    // enable output on gpio (inverting)
    gpio_set_oeover(gpio, 0x1);
    // set output signal for gpio: drive output low
    gpio_set_outover(gpio, 0x2);
}
```

hajimen, 2022-11-10:
It didn't work.

> ```
> void gpio_set_mode_open_drain(uint gpio) {
>     // enable output on gpio (inverting)
>     gpio_set_oeover(gpio, 0x1);
>     // set output signal for gpio: drive output low
>     gpio_set_outover(gpio, 0x2);
> }
> ```

Because the "oe", "output enable", corresponds to SIO: GPIO_OE register.
OEOVER of IOBANK0:GPIOn_CTRL will be for some tricky thing of PIO, not for SIO.
Just use SIO: GPIO_OE, instead of GPIO_OUT (should be L).

DietmarSchwertberger, 2023-02-19:
> Every Push-Pull output as the RP2040 has, is also a Open-Drain-Output.

This is not correct. Push-Pull is not an Open-Drain-Output. You can wire several Open-Drain-Outputs together to one line with a pull-up. Any of the OD outputs can then pull down the signal line to have a wired or. This is used on many opportunitites, e.g. for communication bus or interrupt lines.