raspberrypi/pico-sdk #753: Suggest to change the start up delay time to 2 ms

maowenniao, 2022-03-09:
I tried to build a custom PCB around pico. Try to chose one available crystal is difficult. Due to the official SDK has already limit the Oxsc start up time to 1ms. Most of 2520 packaging size crystals can’t start up in 1 ms with a recommend capacitor circuit. So could the official SDK can adjust the delay time of the function of start up to 2ms. Maybe it could well-matched to almost all kinds of crystals.

lurch, 2022-03-09:
See e.g. https://github.com/raspberrypi/pico-sdk/blob/develop/src/boards/include/boards/adafruit_trinkey_qt2040.h#L18 ?

maowenniao, 2022-03-09:
> See e.g. https://github.com/raspberrypi/pico-sdk/blob/develop/src/boards/include/boards/adafruit_trinkey_qt2040.h#L18 ?

Yeah, I know it can be solved by building a custom uf2 firmware. But I still hope the official SDK can do this change. Because the official SDK is the still the top priority for most of us.

lurch, 2022-03-09:
I'm not sure what it is that you're asking for? The SDK builds UF2s for the Raspberry Pi Pico by default; but if you're building for your own board (which I assume you are, as you mentioned using a different crystal?) then you'll need to build UF2s customised to run on your board, as explained in Appendix D of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

maowenniao, 2022-03-09:
> I'm not sure what it is that you're asking for? The SDK builds UF2s for the Raspberry Pi Pico by default; but if you're building for your own board (which I assume you are, as you mentioned using a [...]

Thanks, got it. So should this variable be changed when I want build a custom UF2 for the micropython on my board? 
https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2040/hardware_regs/include/hardware/regs/xosc.h#L123

lurch, 2022-03-09:
That sounds like a question for the [MicroPython forums](https://forum.micropython.org/viewforum.php?f=21), it's not something that we support ourselves.

maowenniao, 2022-03-09:
Thanks a lot!

> That sounds like a question for the [MicroPython forums](https://forum.micropython.org/viewforum.php?f=21), it's not something that we support ourselves.

closed 2022-03-09 by maowenniao

maowenniao, 2022-03-09:
> The preprocessor value `PICO_XOSC_STARTUP_DELAY_MULTIPLIER` is available as part of the SDK. It's set in a board-specific way for a particular board. We use it for Adafruit boards, because we had similar problems. So for MicroPython you would just set that in the board-specific parameters for your particular board build.

Yeah, get it. Very helpful. Thanks a lot!

dogtopus, 2022-05-01:
While changing the delay per board works on paper (and is probably the most proper way), there is at least 1 hidden problem: custom board makers might have a hard time explaining to the clients why common utilities that seem "board-agnostic" like nuke or even micropython doesn't work on their boards, which then might become a support nightmare. THis risk will always exist unless they also uses the exact same crystal and load capacitor configuration on Pico (which I think is implicitly discouraged since the exact crystal part number is never public and the reference PCB uses a crystal with huge CL which could cause slow startup and I suspect that might be the culprit of this issue).

lurch, 2022-05-04:
> the exact crystal part number is never public

https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#design-files links to https://datasheets.raspberrypi.com/pico/RPi-Pico-R3-PUBLIC-20200119.zip and inside that zipfile there's a `RPI-PICO_PCB-R3_BOM-R3-PUBLIC.xlsx` file which says:
**Reference**: X1
**Part**: 12MHz
**Size**: 3.2x2.5x0.6mm
**Tempco**: 30ppm
**Frequency**: 12MHz
**Manufacturer**: AEL
**Manuf. Part No**: X12M000000S096
**Description**: Xtal, SMT, 3.2x2.5mm, 30ppm, 50R, Fundamental, 10p load, 12MHz

dogtopus, 2022-05-05:
> X12M000000S096

> https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#design-files links to https://datasheets.raspberrypi.com/pico/RPi-Pico-R3-PUBLIC-20200119.zip and inside that [...]

Thanks for this info. Somehow I missed it.