raspberrypi/pico-sdk #755: Multipe versions of vgaboard.h

Memotech-Bill, 2022-03-11:
It would be useful if there were three different versions of the vgaboard,h board definition file for the Pimoroni VGA Demo board:

* vgaboard_serial.h - For when the SD card is not being used. This would not define any of the PICO_SD macros.

* vgaboard_sd.h - For when the SD card is being used. This would not define any of the PICO_DEFAULT_UART macros.

* vgaboard_cut.h - For when the tracks between the Pico and SD Data 1 & 2 have been cut (or the links on the prototype board removed). This would define both the PICO_DEFAULT_UART and PICO_SD macros, but with PICO_DAT_PIN_COUNT=1.

This would make it easier to write code which has to be able to compile for any of these configurations.

lurch, 2022-03-11:
I believe the `vgaboard.h` was added to reflect the VGA Demo Board _example_ design in https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf

If the **Pimoroni** VGA Demo board works differently, then it's up to them to provide appropriate board-definition files.
ping @ZodiusInfuser (who added many of the previous Pimoroni board definition files)

kilograham, 2022-03-12:
You can create your own board header files.

Just set the `PICO_BOARD_HEADER_DIRS` to point to the directory where you keep them, either in your environment, on the cmake command line or at the top of your `CMakeLists.txt`

Memotech-Bill, 2022-03-12:
With the reference board design, there are at least two hardware configurations, J7 links-in or links-out. If the links are in (or on the Pimoroni board the links are ot cut) the user has to decide whether their design will use the SD card, or use the serial port as it is not possible to use both in this configuration. Hence the three board definitions.

I know that I can create my own board definitions, but ideally I would like to create a repository containing a CMake file that will compile for any user selected board. This would require being able to specify a search path for the board files, i.e. first look for any custom board definitions, and if not found there look in the standard board definitions.

Memotech-Bill, 2022-03-12:
I have just re-read the relevant section of the documentation and discovered that PICO_BOARD_HEADER_DIRS is a search path, not a single folder so I can create my own board definition files and have either these or the standard ones used.

I still think that my argument that there should be three different files is valid.

lurch, 2024-06-05:
> I still think that my argument that there should be three different files is valid.

Wouldn't it be simpler to just have a bunch of `#if / #elif / #endif` directives in the existing `vgaboard.h` ? :thinking:

Memotech-Bill, 2024-06-06:
> > I still think that my argument that there should be three different files is valid.
> 
> Wouldn't it be simpler to just have a bunch of `#if / #elif / #endif` directives in the existing `vgaboard.h` ? 🤔

That would work. I don't know whether it would be simpler.

* With three board definitions you just have to specify which board to use.
* With conditionals in `vgaboard.h` you have to both specify that board definition, and set the `#defines` needed to choose the particular configuration.