raspberrypi/pico-sdk #758: Add flash read/write API

yonsch, 2022-03-16:
Currently the flash can be accessed using the two ROM functions `flash_range_program` and `flash_range_erase`. However, there is no proper way to read from the flash, and to write individual bytes. Having these two features would allow users to use parts of the flash as non volatile memory.

**Considered alternatives**
Reading can be implemented using `flash_do_cmd`, though it requires allocating a large tx array to satisfy the API. Writing to arbitrary byte ranges can be implemented using `flash_range_program` with 0xff padding for unaffected bytes. Both can be implemented by writing directly to the SSI. But it would be better to have this done in the SDK and not application code.

lurch, 2022-03-16:
> Writing to arbitrary byte ranges can be implemented using flash_range_program with 0xff padding for unaffected bytes.

How does that work when a flash-address is programmed with e.g. `0x55` and you want to change it to `0xAA` ?

If you want to read/write from the flash as a filesystem, [MicroPython](https://micropython.org/) contains support for that. (although I know that doesn't help if you're a C programmer!)

yonsch, 2022-03-17:
@lurch Thanks, I'll take a deeper look into the micropython implementation, though in a quick glance it seemed that it writes in blocks, not bytes.
As for your question, it can't be done without erasing the flash first, but I'm assuming a higher level file system will take care of that.

lurch, 2022-03-17:
IIRC @kilograham _did_ mention previously that he wanted to add some kind of filesystem support to the SDK eventually, but he's simply been too busy with other stuff (including [this](https://www.raspberrypi.com/news/doom-comes-to-raspberry-pi-pico/) ).

lurch, 2022-03-17:
Aha #531

yonsch, 2022-03-20:
OK, good to know filesystem support is on its way. In my case I already have a filesystem, so I only need a read/write API (I'm writing a Zephyr flash driver).

willeccles, 2022-04-01:
I have been using the flash as non-volatile memory without issue--there's even an example program for this in the pico-examples repo. I just write in full pages, which is slightly inefficient for smaller data, but not enough to make a difference. I can then read arbitrarily from flash using a pointer. See here for an example: https://github.com/raspberrypi/pico-examples/blob/master/flash/program/flash_program.c