raspberrypi/pico-sdk #759: RP2040-E5 fix needs tinyusb_additions added to CMakeLists.txt

trejan, 2022-03-17:
The RP2040-E5 fix needs [TUD_OPT_RP2040_USB_DEVICE_ENUMERATION_FIX](https://github.com/hathach/tinyusb/blob/5b7a01ba79f7a048ddea3b5dedf187746cd21225/src/portable/raspberrypi/rp2040/dcd_rp2040.c#L303) to be set for tinyusb to call rp2040_usb_device_enumeration_fix() during the IRQ handler. TUD_OPT_RP2040_USB_DEVICE_ENUMERATION_FIX is defined by [PICO_RP2040_USB_DEVICE_ENUMERATION_FIX](https://github.com/hathach/tinyusb/blob/5b7a01ba79f7a048ddea3b5dedf187746cd21225/src/portable/raspberrypi/rp2040/rp2040_usb.h#L16) and that is set in the [RP2040 family cmake file](https://github.com/hathach/tinyusb/blob/69ef918021b1c4c97db3bfa41f73b0348ec617c8/hw/bsp/rp2040/family.cmake#L106) as part of the tinyusb_additions library.  The docs/examples don't have tinyusb_additions in the target_link_libraries list or use family_configure_* which adds tinyusb_additions so pico_fix/rp2040_usb_device_enumeration is compiled as part of the SDK but never used by tinyusb.

There is a [thread on the Pi forum](https://forums.raspberrypi.com/viewtopic.php?t=331142) where the RP2040-E5 issue has was causing problems. Explicitly setting TUD_OPT_RP2040_USB_DEVICE_ENUMERATION_FIX was found to workaround the issue so pointed at the fix not being used.

Proposed fix would be to add tinyusb_additions to target_link_libraries.

JamesH65, 2022-03-18:
@kilograham @liamfraser FYI in case you hadn't already seen it.

kilograham, 2022-05-04:
I'm not 100% sure where you are asking for this to be added? What "docs/examples" are you referring to.

Note that the fix is not the default, as it takes control of GPIO 15, and 16 briefly, so you have to know you are allowed to do that.

trejan, 2022-05-04:
The datasheet shows the E5 errata and points to https://github.com/raspberrypi/pico-sdk/tree/master/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c as the workaround. It doesn't say that you need to specifically enable it or how to enable it. Nothing in https://github.com/raspberrypi/pico-examples mentions it either.

What is the approved method for enabling the E5 workaround? Do you add the tinyusb_additions library or manually set the define? If you set a define, which one do you set?

tjko, 2022-05-05:
I had missed the part that this uses GPIO 15/16.  Brief explanation what happens and when would be helpful.

I ran into this "enumeration" issue recently and adding following into CMakeLists.txt seemed to work fine (on a RP2040-B1):
```
target_compile_definitions(targetname PRIVATE PICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1)
```

I'm using GPIO 15 for PWM input and GPIO 16 for PWM output, so presumably I might see short interruption on PWM output, etc? But presumably this would happen only when USB cable is plugged in (or could this happen if there is USB "bus reset" etc?)

kilograham, 2022-05-12:
> I had missed the part that this uses GPIO 15/16. Brief explanation what happens and when would be helpful.
>
> I ran into this "enumeration" issue recently and adding following into CMakeLists.txt [...]

Correct; just adding `PICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1` is all you need to do; I have added info to the errate for the imminent SDK1.3.1. release

actually only GPIO 15 is affected.

it happens on any bus reset.

lurch, 2022-05-18:
See also https://github.com/raspberrypi/pico-examples/pull/210

closed 2022-05-18 by trejan