raspberrypi/pico-sdk #760: Include user_data with UART IRQ callback

ShellAddicted, 2022-03-19:
Please include user_data also in `irq_set_exclusive_handler()`

This links to #71 & #756

kilograham, 2022-03-20:
no; this takes up RAM the user might not need.
You can easily roll your own by keeping a global variable (per IRQ handler); if you are using the same IRQ handler for multiple IRQs, then you can easily make a wrapper handler which calls the shared handler, or even use the current exception number (i.e. the number of the IRQ being handled) to index an array,

closed 2022-03-20 by kilograham

ShellAddicted, 2022-03-22:
@kilograham I opened this issue because it limits the ability to use `irq_set_exclusive_handler()` in an OOP C++ context.
I don't see a way to use a class method as an interrupt handler if it's not possible to pass any argument to it.

Do you agree with this limitation?