raspberrypi/pico-sdk #773: Please don't squash on releases

flaviut, 2022-03-30:
The 1.3.0 release was done by squashing all commits together into one.

This makes is very difficult to find out which commits are in a certain release. I'm generally all for squashing history in PRs and similar, but I would greatly appreciate it if future releases were a merge or fast-forward instead.

aswild, 2022-04-25:
I second this. Squashed releases make commands like "git log 1.3.0..origin/develop" basically useless.

For reference to anyone else who finds this issue, commit 60c5a929ccf9d475e5bd050937af4d599fa92e64 "Remove <b> from mutex doxygen due to PDF generation bug" on the develop branch appears to be what they squash-merged the to master for the 1.3.0 tag, their git trees are identical.

kilograham, 2022-05-04:
agreed

kilograham, 2022-05-10:
anyone have an opinion on whether to retag `60c5a92` as 1.3.1?

kilograham, 2022-05-17:
I created a new merge commit in master `ea79d29c` which i may retag as 1.3.0. this has the actual history and matched `60c5a92` exactly.

`8195d99f` which is currently  branch `sdk1.3.1-candidate` is what should become the new release (building on that)

kilograham, 2022-05-17:
actually i did update the tag and note it in the release notes

closed 2022-05-17 by kilograham

flaviut, 2022-05-17:
Thank you, I appreciate it! I didn't see the earlier comment, but I'm definitely in favor of retagging.

dhalbert, 2022-05-20:
The retagging broke our use of `pico-sdk` as a submodule. https://github.com/circuitpython has a specialized action for updating submodules, but it's possible it might have broken it anyway. We'll advance to 1.3.1.
```
From https://github.com/raspberrypi/pico-sdk
 * branch            2062372d203b372849d573f252cf7c6dc2800c0a -> FETCH_HEAD
 ! [rejected]        1.3.0      -> 1.3.0  (would clobber existing tag)
fatal: run_command returned non-zero status for ports/raspberrypi/sdk
.
make: *** [Makefile:332: fetch-submodules] Error 128
```

kilograham, 2022-05-21:
ah, sorry about that. six of one, half a dozen of the other.. you could just delete the local tag too first

xxc3nsoredxx, 2022-07-10:
> actually i did update the tag and note it in the release notes

@kilograham Please _don't_ change existing tags in the future. It breaks anything using tags to download the source tree for a specific version, and relying on each download being identical. [For example, Linux distro packaging](https://github.com/gentoo-mirror/unc3nsored/commit/f0d8ee58748d07cd7f5502f40b37a4df11bd7e7a)

Also, from the [`git-tag` docs](https://git-scm.com/docs/git-tag#_discussion) (emphasis mine):
> But if you have pushed things out (or others could just read your repository directly), then others will have already seen the old tag. In that case you can do one of two things:
>
> 1. The sane thing. Just admit you screwed up, and use a different name. Others have already seen one tag-name, and if you keep the same name, you may be in the situation that two people both have "version X", but they actually have different "X"'s. So just call it "X.1" and be done with it.
>
> 2. _**The insane thing. You really want to call the new version "X" too, even though others have already seen the old one. So just use git tag -f again, as if you hadn’t already published the old one.**_
>
> ... If somebody got a release tag from you, you cannot just change the tag for them by updating your own one. This is a big security issue, in that people MUST be able to trust their tag-names.