raspberrypi/pico-sdk #777: "target halted..." suppresses target output when Semihosting enabled

a-gavin, 2022-04-05:
### Issue Description
In debugging code for a Semihosting-enabled Pico using another Pico flashed with Picoprobe, **a problem I have run into is the overload of ``target halted...`` output visually suppressing Semihosting output from the target device**. The actual Semihosting output appears interleaved with other output in OpenOCD output making it difficult to read. This output only periodically appears in GDB output.

I realize this is not really a Pico-SDK issue, but I am posting here as my Google-fu here is lacking, I'm sure someone may have some good pointers, and I am happy to contribute to documentation for using Semihosting with the Pico-SDK (potentially Section 6 of the [*Getting started with Raspberry Pi Pico*](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)?).

For context, I have followed the *Getting started with Raspberry Pi Pico* guide to setup the debug probe and build system for the target device. Example target device code, CMakeLists.txt, and commands invoked are listed at the end of the issue.

The actual looks as below. The expected output from the target device is the output of a forever while loop printing ``"Spinning\n"`` once per second. As you can see below, each remnants of this are visible as the beginning character of every other line. My goal here is to configure whatever is necessary to remove the ``target halted..`` output so I can see the actual target device output. Any help is appreciated!

**OpenOCD output**
```
Starget halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
ptarget halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
itarget halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
ntarget halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
ntarget halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
itarget halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
ntarget halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
```
### Code
**test.c**
```test.c
#include <stdio.h>
#include <pico/stdlib.h>

int main() {
    stdio_init_all();

    while (true) {
        printf("Spinning\n");
        sleep_ms(1000);
    }
}
```
**CMakeLists.txt**
```CMakeLists.txt
cmake_minimum_required(VERSION 3.13)

include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)

project(test C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

if (PICO_SDK_VERSION_STRING VERSION_LESS "1.3.0")
    message(FATAL_ERROR "Raspberry Pi Pico SDK version 1.3.0 (or later) required. Your version is ${PICO_SDK_VERSION_STRING}")
endif()

# Initialize the SDK
pico_sdk_init()

add_compile_options(-Wall
        -Wno-format          # int != int32_t as far as the compiler is concerned because gcc has int32_t as long int
        -Wno-unused-function # we have some for the docs that aren't called
        -Wno-maybe-uninitialized
        -g
)

add_executable(test
        test.c
)

# pull in common dependencies
target_link_libraries(test pico_stdlib)

# tell the pico library that you will be using usb serial and not an actual uart on the 
# processor
pico_enable_stdio_usb(test 1)
pico_enable_stdio_uart(test 0)
pico_enable_stdio_semihosting(test 1)

# create map/bin/hex file etc.
pico_add_extra_outputs(test )

add_custom_command(TARGET test POST_BUILD COMMAND arm-none-eabi-size test.elf)
```
**gdbinit**
```
target remote localhost:3333
monitor arm semihosting enable
monitor reset halt
```
### Commands Invoked (output removed for brevity)
**Shell 1**
```sh
openocd -f interface/picoprobe.cfg -f target/rp2040.cfg
```
**Shell 2**
```sh
mkdir build
cd build 
cmake ..
make -j8
gdb-multiarch --init-command=../gdbinit test.elf
```
**In GDB**
```
load
c
```

a-gavin, 2022-04-05:
As an addendum, I should also not that I am using the following:
Pico-SDK: v1.3.0
Picoprobe: Commit ``164eaa5b80295ff9ef3a8ebf8949d5db4699f904`` on remote https://github.com/raspberrypi/picoprobe.git
OpenOCD: Commit ``df76ec7edee9ebb9be86e4cff7479da642b0e8df`` on remote https://github.com/raspberrypi/openocd.git

lurch, 2022-04-05:
> pico_enable_stdio_usb(test 1)
pico_enable_stdio_uart(test 0)
pico_enable_stdio_semihosting(test 1)

Why are you trying to use two different stdio modes at the same time? Is that deliberate? Could it be why you're seeing interleaved output? :shrug:

a-gavin, 2022-04-05:
> Why are you trying to use two different stdio modes at the same time? Is that deliberate? Could it be why you're seeing interleaved output? 🤷

I hadn’t considered that. Should these modes be mutually exclusive?

lurch, 2022-04-05:
I can't give any further advice as I've never used semi-hosting, but maybe @kilograham or @liamfraser will come along later with some suggestions.

But I _did_ experiment with using both pico_enable_stdio_usb and pico_enable_stdio_uart at the same time a while ago, and IIRC that worked fine.

a-gavin, 2022-04-06:
Just taking a cursory peek at the code for the stdio output function eventually called by ``printf`` [here](https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2_common/pico_stdio/stdio.c#L106) it appears that enabling multiple stdio modes should work.

a-gavin, 2022-04-06:
Did a bit more digging this evening and found the printout I was seeing [here](https://github.com/raspberrypi/openocd/blob/610f137d2b91b794dc9a8999b1e6631a99fcddc7/src/target/armv7m.c#L696) in OpenOCD. Adding the following to the end of  my ``gdbinit`` file eliminated the printout but did not print the target device output to GDB: ``monitor debug_level -2`` Other levels of OpenOCD output configuration can be found [in this OpenOCD issue](https://github.com/riscv/riscv-openocd/issues/193).

@lurch I was able to get this to work with other stdio modes enabled (usb and uart). Thank you for your help!

Closing this issue as is not Pico-SDK related at this point.

closed 2022-04-06 by a-gavin