raspberrypi/pico-sdk #780: pico-sdk and tinyusb fail to build with GCC 11.2

lurch, 2022-04-07:
I'm not sure exactly where the problem is, so pinging @kilograham @liamfraser and @hathach 

If I build _just_ `pico-sdk` with GCC 11.2 from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads then it builds fine, but if I also clone the tinyusb sumbodule, then when I try to build it fails with:
```
[ 75%] Building C object test/kitchen_sink/CMakeFiles/kitchen_sink_extra_stdio.dir/__/__/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj
/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c: In function 'rp2040_usb_init':
/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:61:3: error: 'memset' offset [0, 155] is out of the bounds [0, 0] [-Werror=array-bounds]
   61 |   memset(usb_hw, 0, sizeof(*usb_hw));
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c:62:3: error: 'memset' offset [0, 4095] is out of the bounds [0, 0] [-Werror=array-bounds]
   62 |   memset(usb_dpram, 0, sizeof(*usb_dpram));
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [test/kitchen_sink/CMakeFiles/kitchen_sink_extra_stdio.dir/build.make:781: test/kitchen_sink/CMakeFiles/kitchen_sink_extra_stdio.dir/__/__/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:2069: test/kitchen_sink/CMakeFiles/kitchen_sink_extra_stdio.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
:cry: 

To reproduce:
(I'm doing everything inside docker just to rule out any other part of the system interfering with things)
```bash
sudo docker run -it ubuntu:20.04
```
and then inside docker:
```bash
apt update
apt install -y build-essential git python3 cmake wget
wget https://developer.arm.com/-/media/Files/downloads/gnu/11.2-2022.02/binrel/gcc-arm-11.2-2022.02-x86_64-arm-none-eabi.tar.xz
tar xf gcc-arm-11.2-2022.02-x86_64-arm-none-eabi.tar.xz
git clone https://github.com/raspberrypi/pico-sdk.git
cd pico-sdk
mkdir build.nousb
cd build.nousb
PATH=../../gcc-arm-11.2-2022.02-x86_64-arm-none-eabi/bin:$PATH cmake ..
make
# this 'make' all works fine, and builds to 100%
cd ..
git submodule init
git submodule update
mkdir build.usb
cd build.usb
PATH=../../gcc-arm-11.2-2022.02-x86_64-arm-none-eabi/bin:$PATH cmake ..
make
# this 'make' fails with the memset error detailed above
```

I'm happy to provide any other info and/or do any other testing that might be needed.

kilograham, 2022-04-07:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578 (haven't read it all yet)

kilograham, 2022-04-07:
ah; fixed in GCC 11.3; we can probably suppress it in 11.2 if we really want as we already have code to suppress tinyusb warnings

lurch, 2022-04-07:
> we can probably suppress it in 11.2 if we really want

Probably worth doing - who knows how long it'll be until ARM provide an updated "Arm GNU Toolchain" ? :shrug:

JamesH65, 2022-04-07:
> https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578 (haven't read it all yet)

That's quite a thread. Someone wants to change 40 years of embedded programming practice by warning about it, and get everyone to change their code to get rid of the warning. Only about a billion lines of code. Nice. At least sense was seen.

kilograham, 2022-04-07:
> That's quite a thread. Someone wants to change 40 years of embedded programming practice by warning about it, and get everyone to change their code to get rid of the warning. Only about a billion [...]

yeah, i'm surprised ARM built an embedded toolchain (actually i guess this is now not just an embedded toolchain - i was going to the old page which is why i hadn't seen 11.2) which is likely to give warnings on all embedded projects (I'm surprised they didn't hit it in testing)

geurtv, 2022-04-07:
FYI: the latest tinyusb already surpresses -Warray-bounds with a #pragma in code, but -Wstringop-overflow isn't surpressed (yet).

beroset, 2022-05-04:
For what it's worth, I added this ugly hack to my copy which adds no actual code bytes, but suppresses the warning:
```
static void memset_nowarn(void *d, int c, size_t len)
{
    asm("" : "+r"(d));
    memset(d, c, len);
}
```
Example on compiler explorer:  https://godbolt.org/z/9xYGjdEx5

kilograham, 2022-05-11:
merged into develop

closed 2022-05-11 by kilograham

lurch, 2022-05-12:
Just for the record, the instructions in my first post above now all work on the `develop` branch of pico-sdk in docker containers for all combinations of:
 * Ubuntu 20.04 / 22.04
 * gcc-arm-none-eabi 11.2 / the OSes default-version
 * with / without `tinyusb` submodule

and for both `pico-sdk` and `pico-examples`. :tada:

beroset, 2022-05-12:
Confirming as well that the `develop` branch now compiles cleanly with Fedora 35.