raspberrypi/pico-sdk #781: can't find stdlib

grungy01, 2022-04-07:
i've followed several tutorial  in order to install pico's sdk ( after 2 failed installations on windows i gave up and switch on linux to trie), i'm trying to launch the hello world project from the pico-examples in visual studio but i get an error (can't find 'pico/stdlib.h') : 

`main] Configuring folder: pico-examples 
[proc] Executing command: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/arm-none-eabi-g++ -S/home/grungy/rpi-pico/pico-examples -B/home/grungy/rpi-pico/pico-examples/build -G "Unix Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Configuring incomplete, errors occurred!
[cmake] Using PICO_SDK_PATH from environment ('/rpi-pico/pico-sdk')
[cmake] CMake Error at pico_sdk_import.cmake:52 (message):
[cmake]   Directory '/rpi-pico/pico-sdk' not found
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:4 (include)`

ps : path is configured

kilograham, 2022-04-07:
the individual examples are not projects in their own right... you should open the top level project

kilograham, 2022-04-07:
Equally the errors

```
[cmake] Using PICO_SDK_PATH from environment ('/rpi-pico/pico-sdk') 
[cmake] CMake Error at pico_sdk_import.cmake:52 (message): [cmake] Directory '/rpi-pico/pico-sdk' not found 
```

seems like it would be a problem.

lurch, 2022-04-07:
> i've followed several tutorial in order to install pico's sdk ( after 2 failed installations on windows i gave up and switch on linux to trie),

If you can point out where things went wrong when carefully following the instructions in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf that'd be really useful. Obviously any third-party tutorials are beyond our control!

> [cmake] Using PICO_SDK_PATH from environment ('/rpi-pico/pico-sdk') 

Yeah it looks like you've done `export PICO_SDK_PATH=/rpi-pico/pico-sdk` when you actually want to do `export PICO_SDK_PATH=/home/grungy/rpi-pico/pico-sdk` ? (or `export PICO_SDK_PATH=$HOME/rpi-pico/pico-sdk`)

grungy01, 2022-04-08:
@lurch  i"ve checked the command that i've launched it was export PICO_SDK_PATH=$HOME/rpi-pico/pico-sdk.
I reinstalled the sdk following the doc, installed the additional libraries for ubuntu-Debian, all like the first I met no error message. I created the build directory and then opened visual studio, I open the pico-examples directory under vs studio, I open the hello_usb.c file and there I still have the same message: no file corresponding to "pico/stdlib.h". 

 I did a search in the sdk, the file is located in /home/grungy/pico/pico-sdk/src/common/pico_stdlib/include/pico/stdlib.h. 
and my path is pointing now on /home/grungy/pico/pico-sdk/

lurch, 2022-04-08:
> I still have the same message: no file corresponding to "pico/stdlib.h"

That might just be a warning from the syntax-highlighter? See e.g. https://github.com/raspberrypi/pico-feedback/issues/144 for a possible workaround. (unfortunately I don't have time to look at that in detail right now)

But if you ignore that warning, does the code actually build?

(although I'm slightly confused why your first comment mentions `/home/grungy/rpi-pico/`, and in your most recent comment you're talking about `/home/grungy/pico/` ?)

kilograham, 2022-04-08:
i don't use visual studio, but it sounds like you need to clear the CMake cache (i assume there is a menu somewhere which does that)

kilograham, 2022-05-09:
help issue, and haven't heard in a while, so closing

closed 2022-05-09 by kilograham