raspberrypi/pico-sdk #782: access pico's ram through swd pins using another pico

Friday8229, 2022-04-08:
Okay so I'm trying to write some data to a pi pico using another pico. I know i can use a custom bootloader to achieve this but i want to write a program to a fresh pico (that is not running anything) using another pico.

This can be achieved using openocd and a raspberry pi or another Pico as picoprobe but i don't want to have a Linux machine.

Basically i want to program a pi pico from another pico.

Any suggestions how should I proceed?

lurch, 2022-04-08:
> Any suggestions how should I proceed?

Sounds like you'll need to study the picoprobe sourcecode to see how it works, and then adapt it to suit your needs?

@kilograham I guess this ought to be moved to the picoprobe repo?

Friday8229, 2022-04-08:
Thanks for the quick reply, sure I'll check out picoprobe source code. I wanna know if it's even possible to write code like that ?

lurch, 2022-04-08:
I've not looked at the [picoprobe](https://github.com/raspberrypi/picoprobe/) sourcecode myself, but my (vague) understanding is that the host PC sends data to picoprobe, and picoprobe then writes that data to the secondary Pico using SWD. So I guess you "just" need to modify where picoprobe gets the data from? :shrug:

Friday8229, 2022-04-08:
That's what I thought... I'll try to Modify the Picoprobe code, tho i think it's not an easy task to eliminate the pc and do everything on the picoprobe.

closed 2022-04-10 by Friday8229