raspberrypi/pico-sdk #786: gpio_get() doesn't work for high numbered channels

Gavin-Perry, 2022-04-17:
gpio_get() doesn't seem to work for channels 26-28 and possibly others above 16?
example

 gpio_set_dir(27, GPIO_IN);
  gpio_pull_up(27); // set to 1 or
 // gpio_pull_down(27); // set to 0
// Try it...
  my27Val = gpio_get(27);
// or
if (gpio_get(26)) ChanGroup = 1;
// ALWAYS returns 0

I used a work around using analogRead() which works so it's not a hardware issue.
if (analogRead(27)>400) my27Val = 1;  else my27Val=0; // 
// I suppose just would work
my27Bool = (analogRead(27)>400);

lurch, 2022-04-17:
If you're mixing up analog code and digital code, then I _think_ setting an "ADC-capable GPIO" (i.e. GPIO26-GPIO29) into analog mode disconnects it from the digital controls?
Can you paste a full example program that demonstrates the problem please, rather than just a short snippet?

geurtv, 2022-04-21:
Seems like those pins are already set to some alternate function (by SDK or bootloader?) or something. I don't know what or why but when I run your example I get the same results so pull-up doesn't work. Anyway, you need to start with `gpio_init()`, then it works for me:

```c
gpio_init(26)
gpio_set_dir(26, GPIO_IN);
gpio_pull_up(26);
... etc
```

lurch, 2022-04-21:
> Seems like those pins are already set to some alternate function (by SDK or bootloader?) or something. I don't know what or why

In Section 2.3.1.2 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf it says "To drive a pin with the SIO’s GPIO registers, the GPIO multiplexer for this pin must first be configured to select the SIO GPIO function." and I believe that's what the `gpio_init()` function does. Looking up the `GPIO0_CTRL` register (in Table 295) shows that the `FUNCSEL` field defaults to NULL, i.e. each external GPIO pin is "non-functioning" until assigned to a specific function-select.

Also, once you have the GPIOs and pull-ups properly configured, I guess your gpio-reading code could (optionally) be reduced down to something like:
```
ChanGroup = (gpio_get_all() & (1<<26 | 1<<27 | 1<<28)) >> 26;
```
to read all three GPIOs into a single 3-bit value with just one function-call.

lurch, 2022-05-18:
I guess this can be closed now?

closed 2022-05-26 by Gavin-Perry