raspberrypi/pico-sdk #787: reading hardware PWM CC value

Gavin-Perry, 2022-04-17:
I'm using hardware PWM for LED dimming I'd like to read back the counter compare (e.g. ch0_cc etc without calculating where the registers are located.) I want to slew from one given DMX value to the next in the 16bit domain as otherwise LEDs flicker (too fast!)

function pwm_set_gpio_level(pin,ccVal); // works fine but there is no complimentary
function lastCCVal =pwm_get_gpio_level(pin);
other registers have a read function even
  pwm_get_counter (uint slice_num) which tells the current count not the compare setting

Seems like it would be easy and obvious to do.  If not then can you supply the address calculation (or the helper function/macro needed)

lurch, 2022-04-17:
I guess the double-buffering mentioned in section 4.5.2.3. of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf might mean that (depending on timings) you only get back the "latched" value of CC, rather than the value currently being used by the hardware? But if you're happy with (possibly) only reading the latched values, then I believe https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_pwm/include/hardware/pwm.h#L302 and https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_pwm/include/hardware/pwm.h#L255 show the calculations you're looking for?

Gavin-Perry, 2022-04-20:
I use pwm_set_gpio_level(uint gpio, uint16_t level)  so just thought pwm_get_gpio_level(uint gpio, uint16_t level) would be handy. Easy work around is to save a copy of what I had just set as lastLevel.
I need it because I want to slew to the next level with multiple calls rather the jumping.
getting the current count with  pwm_get_counter (uint slice_num) lets me know when to push out the next one.
(Wait until it is close to TOP)

closed 2022-04-25 by Gavin-Perry