raspberrypi/pico-sdk #791: USB Host Mass Storage Device

BabumonGopi, 2022-04-22:
Any working program for Mass Storage Device working with pico-sdk ? I had tried to connect with TinyUSB, but it will not working for pico. The following error is coming.

assertion "ep" failed: file "/Users/babumongopi/r-pi/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line 492, function: hcd_edpt_xfer

Any solution ?

mimoccc, 2022-04-22:
The best examples are in tinyusb library i think.
Look up for lun msc code there.
That example works perfectly.

mimoccc, 2022-04-22:
Here: https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_msc
And here: https://github.com/hathach/tinyusb/tree/master/examples/device/msc_dual_lun

lurch, 2022-04-22:
ping @hathach

hathach, 2022-04-22:
bulk endpoint for host on rp2040 is not supported yet. Please wait until it is implemented

BabumonGopi, 2022-04-22:
Is it the issue of RP2040 microcontroller itself or TinyUSB library ?

Babumon Gopi

Poornasree Equipments
www.poornasree.com

hathach, 2022-04-23:
there is no issue with either rp2040 or TinyUSB. There is only issue with time or lack of it.

kilograham, 2022-05-09:
Bumping to 1.4.0 to review when we update TinyUSB

kilograham, 2023-01-20:
@liamfraser can you move this into 1.5.0 if it is fixed with latest TinyUSB

kilograham, 2023-02-11:
see `tinyusb_host_msc_file_explorer` in pico-examples (example from TinyUSB [here](https://github.com/hathach/tinyusb/tree/master/examples/host/msc_file_explorer)

closed 2023-02-11 by kilograham