raspberrypi/pico-sdk #803: pioasm: please keep the labels

smurfix, 2022-05-04:
pioasm replaces labels with line numbers. This is unfortunate. Please don't do that.

(a) Micropython can deal with textual labels, no problem at all.
(b) For C I'd like to be able to refer to a specific instruction so that I can send a jump command to PIO. Currently I need to look at the generated code and/or count instructions, and woe is me when that location changes.

lurch, 2022-05-04:
Section 3.3.5 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf says "A PUBLIC label is exposed to the user code in the same way as a PUBLIC .define."
Is that what you're looking for, or are you requesting something different?

kilograham, 2022-07-11:
This is quite a change to fix... is the PUBLIC not a workaround?