raspberrypi/pico-sdk #804: Feature request: pioasm: multiple programs

smurfix, 2022-05-04:
Let's say I want to use the PIO for something UART-like. Now obviously this requires a send program and a receive program. I want to pack both into one PIO device because they're both small enough.

My problem is that there seems to be no clean way to do that: pioasm doesn't tell me how long the first program is, thus I have no way to set the `.offset` for the second program correctly.

lurch, 2022-05-04:
Have a look at some of the examples in https://github.com/raspberrypi/pico-examples/tree/master/pio/
https://github.com/raspberrypi/pico-examples/tree/master/pio/ir_nec/nec_transmit_library is an example that uses multiple PIO programs (and there might be others if you dig around? :man_shrugging: )

lurch, 2022-05-04:
And I've not tried it myself, but you might want to look at https://github.com/harrywalsh/pico-hw_and_pio-uart-gridge/tree/HW_and_pio_uarts