raspberrypi/pico-sdk #812: i2c_write_raw_blocking needs nostop parameter

sigmafx, 2022-05-05:
Whilst writing to an I2C EEPROM it's necessary to write eeprom memory address followed by the data to write without a STOP between the address and the data. A typical write function would accept a memory address and data as separate parameters. This is best written to the EEPROM in 2 API calls:
1st: Sends device address and eeprom memory address
2nd: Sends data, followed by a STOP

The only achievable way is using the following APIs:
```
        // nostop is true so STOP isn't sent allowing data bytes to follow immediately
        i2c_write_blocking(m_instance, m_devAddr, &address, 1, true);
        i2c_write_raw_blocking(m_instance, &data, sizeof(data));
```
However, `i2c_write_raw_blocking` does not write a STOP.

It would be helpful to include the `nostop` paramter to `i2c_write_raw_blocking` function:
```
static inline void i2c_write_raw_blocking(i2c_inst_t *i2c, const uint8_t *src, size_t len, bool nostop) {
    for (size_t i = 0; i < len; ++i) {
        bool last = i == len - 1;

        // TODO NACK or STOP on end?
        while (!i2c_get_write_available(i2c))
            tight_loop_contents();
        i2c_get_hw(i2c)->data_cmd =
            bool_to_bit(last && !nostop) << I2C_IC_DATA_CMD_STOP_LSB |
            *src++;
    }
}
```
There's already a comment in there about this.

kaimac1, 2024-03-07:
this is also very useful for driving OLED displays, where the frame data needs to be preceded by a single 0x40 byte.

@sigmafx 's function needs the following after the for loop to wait for the STOP bit to actually be generated:

    while (!(i2c->hw->raw_intr_stat & I2C_IC_RAW_INTR_STAT_STOP_DET_BITS)) tight_loop_contents();

lurch, 2024-09-01:
Does #1495 fix this, or is that a different problem entirely?

peterharperuk, 2025-03-03:
If I'm not mistaken, you can do this now with the following functions.

```
/*! \brief Attempt to write specified number of bytes to address, blocking
 *  \ingroup hardware_i2c
 *
 * \param i2c Either \ref i2c0 or \ref i2c1
 * \param addr 7-bit address of device to write to
 * \param src Pointer to data to send
 * \param len Length of data in bytes to send
 * \param nostop  If true, master retains control of the bus at the end of the transfer (no Stop is issued),
 *           and the next transfer will begin with a Restart rather than a Start.
 * \return Number of bytes written, or PICO_ERROR_GENERIC if address not acknowledged, no device present.
 */
int i2c_write_blocking(i2c_inst_t *i2c, uint8_t addr, const uint8_t *src, size_t len, bool nostop);

/*! \brief Attempt to write specified number of bytes to address, blocking in burst mode
 *  \ingroup hardware_i2c
 *
 * This version of the function will not issue a stop and will not restart on the next write.
 * This allows you to write consecutive bytes of data without having to resend a stop bit and
 * (for example) without having to send address byte(s) repeatedly
 *
 * \param i2c Either \ref i2c0 or \ref i2c1
 * \param addr 7-bit address of device to read from
 * \param dst Pointer to buffer to receive data
 * \param len Length of data in bytes to receive
 * \return Number of bytes read, or PICO_ERROR_GENERIC if address not acknowledged or no device present.
 */
int i2c_write_burst_blocking(i2c_inst_t *i2c, uint8_t addr, const uint8_t *src, size_t len);
```

closed 2025-03-03 by peterharperuk