raspberrypi/pico-sdk #820: Upgrade TinyUSB

kilograham, 2022-05-10:
This will land in 1.4.0 not 1.3.x as there may be minor incompatibilities (if major we should consider 2.x)

At this point we should undo #819

kilograham, 2022-12-28:
see also #1044

lurch, 2023-02-04:
I guess when this does happen, we'll also want to update the USB examples in [pico-examples](https://github.com/raspberrypi/pico-examples) too? (and also update the pico-examples README.md to mention any new USB examples that get added)

kilograham, 2023-02-06:
merged into develop

closed 2023-02-06 by kilograham

lurch, 2023-02-12:
> (and also update the pico-examples README.md to mention any new USB examples that get added)

**bump** :wink: 

In #791 @kilograham pointed to the `tinyusb_host_msc_file_explorer` example, but that's not mentioned at https://github.com/raspberrypi/pico-examples#usb-host even though it is listed at https://github.com/hathach/tinyusb/tree/master/examples/host . In fact it looks like _all_ the USB host examples there _do_ include `mcu:RP2040` in their `only.txt` files?

kilograham, 2023-02-12:
well, i didn't re-read all the comments as the title was quite explicit ;-)