raspberrypi/pico-sdk #838: Allow use of pico_stdio_usb with explicit TinyUSB use

kilograham, 2022-05-23:
Currently pico_stdio_usb requires that the user not be using tinyusb, as it exposes its own USB descriptor(s) and runs its own polling loop. 

The majority of pico_stdio_usb would be fine if the user has defined their own CDC device and it would probably be useful to allow that.

lurch, 2022-05-23:
Just out of curiosity, would that interfere with the picotool-reset-over-CDC mechanism?

kilograham, 2022-05-24:
It wouldn't interfere with the reset over CDC mechanism, but that is not what picotool uses. It uses a separate interface, which the user would have to add separately (so the short answer is probably yes)

kilograham, 2022-05-24:
(The CDC mechanism is to set the baud rate, however using that from the host requires you to know how to translate a USB id into a higher level comms device you can talk to which is non-trivial and different on all platforms)

lurch, 2022-05-24:
> It uses a separate interface...

Right, that's what I was thinking of, I guess my mention of CDC must have been a mis-remembering.

kilograham, 2022-06-20:
merged into develop

popy2k14, 2022-11-23:
Hey guys,

Running tiny USB in device mode as hid keyboard and pico tool doesn't work. I am using no CDC in tiny USB. 

When I get it correctly, this should be possible now. Read your thread here and the commits but just don't what to configure to get CDC for pico tool with standard USB lib alongside tiny USB.

Any hints?

kilograham, 2022-11-23:
> Hey guys,
>
> Running tiny USB in device mode as hid keyboard and pico tool doesn't work. I am using no CDC in tiny USB.
>
> When I get it correctly, this should be possible now. Read your thread here and [...]

um CDC != picotool ... you cannot run the picotool interface alongside your app (it is a separate thing int the bootrom)

popy2k14, 2022-11-25:
Added CDC to TinyUSB descriptors and copied and adapted the function:

```
// Support for default BOOTSEL reset by changing baud rate
void tud_cdc_line_coding_cb(__unused uint8_t itf, cdc_line_coding_t const* p_line_coding) {
    if (p_line_coding->bit_rate == 110) {    
        //board_led_write(1);
        reset_usb_boot(gpio_mask, 0);
    }
}
```

Would be nice if this function from sdk reset interface does'nt get's defined out when using TinyUSB.

popy2k14, 2022-11-25:
Done, thx