raspberrypi/pico-sdk #840: Tests fail when using gcc 12

joshuag1000, 2022-05-30:
I ran `cmake .. && make` in a build directory on the pico-sdk using gcc 12 and the kitchen_sink test failed.
I'm not sure if I have configured something incorrectly or not but i believe it is caused by the latest version of gcc.
I have attached the build log below so you can see where it went wrong.
``` 
PICO_SDK_PATH is MyPath/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from MyPath/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.10 (found version "3.10.4") found components: Interpreter 
TinyUSB available at MyPath/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
-- Configuring done
-- Generating done
-- Build files have been written to: MyPath/pico-sdk/build
Scanning dependencies of target bs2_default
[  1%] Building ASM object src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  1%] Linking ASM executable bs2_default.elf
[  1%] Built target bs2_default
[  1%] Creating directories for 'ELF2UF2Build'
[  1%] No download step for 'ELF2UF2Build'
[  1%] No update step for 'ELF2UF2Build'
[  1%] No patch step for 'ELF2UF2Build'
[  1%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: MyPath/pico-sdk/build/elf2uf2
[  1%] Performing build step for 'ELF2UF2Build'
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[  1%] No install step for 'ELF2UF2Build'
[  1%] Completed 'ELF2UF2Build'
[  1%] Built target ELF2UF2Build
[  1%] Generating bs2_default.bin
[  1%] Generating bs2_default_padded_checksummed.S
[  1%] Built target bs2_default_padded_checksummed_asm
Scanning dependencies of target pico_stdlib_test
[  7%] Built target pico_stdlib_test
Scanning dependencies of target pico_time_test
[ 13%] Built target pico_time_test
Scanning dependencies of target pico_divider_test
[ 19%] Built target pico_divider_test
Scanning dependencies of target pico_divider_nesting_test_with_dirty_check
[ 25%] Built target pico_divider_nesting_test_with_dirty_check
Scanning dependencies of target pico_divider_nesting_test_with_disable_irq
[ 31%] Built target pico_divider_nesting_test_with_disable_irq
Scanning dependencies of target pico_float_test
[ 37%] Built target pico_float_test
Scanning dependencies of target pico_double_test
[ 43%] Built target pico_double_test
Scanning dependencies of target kitchen_sink
[... 167 lines cut ...]
MyPath/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:120:59: error: array subscript 0 is outside array bounds of 'uint16_t[0]' {aka 'short unsigned int[]'} [-Werror=array-bounds]
  120 | #define rom_hword_as_ptr(rom_address) (void *)(uintptr_t)(*(uint16_t *)rom_address)
      |                                                          ~^~~~~~~~~~~~~~~~~~~~~~~~~
MyPath/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:130:41: note: in expansion of macro 'rom_hword_as_ptr'
  130 |     uint16_t *func_table = (uint16_t *) rom_hword_as_ptr(0x14);
      |                                         ^~~~~~~~~~~~~~~~
In function 'rom_func_lookup_inline',
    inlined from 'flash_do_cmd' at MyPath/pico-sdk/src/rp2_common/hardware_flash/flash.c:127:76:
MyPath/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:120:59: error: array subscript 0 is outside array bounds of 'uint16_t[0]' {aka 'short unsigned int[]'} [-Werror=array-bounds]
  120 | #define rom_hword_as_ptr(rom_address) (void *)(uintptr_t)(*(uint16_t *)rom_address)
      |                                                          ~^~~~~~~~~~~~~~~~~~~~~~~~~
MyPath/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:129:66: note: in expansion of macro 'rom_hword_as_ptr'
  129 |     rom_table_lookup_fn rom_table_lookup = (rom_table_lookup_fn) rom_hword_as_ptr(0x18);
      |                                                                  ^~~~~~~~~~~~~~~~
MyPath/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:120:59: error: array subscript 0 is outside array bounds of 'uint16_t[0]' {aka 'short unsigned int[]'} [-Werror=array-bounds]
  120 | #define rom_hword_as_ptr(rom_address) (void *)(uintptr_t)(*(uint16_t *)rom_address)
      |                                                          ~^~~~~~~~~~~~~~~~~~~~~~~~~
MyPath/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:130:41: note: in expansion of macro 'rom_hword_as_ptr'
  130 |     uint16_t *func_table = (uint16_t *) rom_hword_as_ptr(0x14);
      |                                         ^~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [test/kitchen_sink/CMakeFiles/kitchen_sink.dir/build.make:438: test/kitchen_sink/CMakeFiles/kitchen_sink.dir/__/__/src/rp2_common/hardware_flash/flash.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1796: test/kitchen_sink/CMakeFiles/kitchen_sink.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

kilograham, 2022-05-30:
See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578

it seems they may not have included the patch in 12.1 which is dumb.

Actually, i'm still surprised this doesn't work with the SDK 1.3.1 - are you sure you have the latest?

kilograham, 2022-05-30:
oh, no, actually this looks like _another new_ warning!

joshuag1000, 2022-05-30:
> See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578
> 
> it seems they may not have included the patch in 12.1 which is dumb.
> 
> Actually, i'm still surprised this doesn't work with the SDK 1.3.1 - are you sure you have the latest?

yeah downloaded it fresh today.

kilograham, 2022-05-31:
i don't see a GCC 12 from ARM directly yet, so I guess this came with some OS?

joshuag1000, 2022-05-31:
I use Arch Linux (x86_64) which ships GCC 12

kilograham, 2022-05-31:
I haven't tested it on GCC 12 myself, but you can try #842

kilograham, 2022-05-31:
hmm; that's odd; for sanity can you comment out the ifdef in `src/rp2_common/pico_bootrom/include/pico/bootrom.h`

`#if defined(__GNUC__) && (__GNUC__ >= 12)`

joshuag1000, 2022-05-31:
sure just let me do that now

kilograham, 2022-05-31:
That is weird too (not sure how it thinks casting from unsigned to unsigned can change the sign of anything!); we may be getting to the point where I will have to try it myself, but could you add 

static_assert(sizeof(uint)==4, "uint should be 4 bytes");

somewhere at the top of rtc.c

joshuag1000, 2022-05-31:
> That is weird too (not sure how it thinks casting from unsigned to unsigned can change the sign of anything!); we may be getting to the point where I will have to try it myself, but could you add
> 
> static_assert(sizeof(uint)==4, "uint should be 4 bytes");
> 
> somewhere at the top of rtc.c

im wondering if this is perhaps an issue with my version of gcc as i just booted up my fedora laptop which claims to use gcc 12.1.1 (which confuses me a little because i dont think that version exists) and the tests ran fine so im going to do some more checks locally to see if my version of gcc is the cause.

joshuag1000, 2022-05-31:
> That is weird too (not sure how it thinks casting from unsigned to unsigned can change the sign of anything!); we may be getting to the point where I will have to try it myself, but could you add
> 
> static_assert(sizeof(uint)==4, "uint should be 4 bytes");
> 
> somewhere at the top of rtc.c

i will try this and see if it addresses the issue

kilograham, 2022-05-31:
Yeah that won't fix it, but I was wondering too if your compiler was a bit weird (seems like it might have the wrong int sizes)

joshuag1000, 2022-05-31:
> > That is weird too (not sure how it thinks casting from unsigned to unsigned can change the sign of anything!); we may be getting to the point where I will have to try it myself, but could you add
> > static_assert(sizeof(uint)==4, "uint should be 4 bytes");
> > somewhere at the top of rtc.c
> 
> im wondering if this is perhaps an issue with my version of gcc as i just booted up my fedora laptop which claims to use gcc 12.1.1 (which confuses me a little because i dont think that version exists) and the tests ran fine so im going to do some more checks locally to see if my version of gcc is the cause.
> 
> Edit: This could just be fedora changing the version numbers as they apply patches or something but still strange how it builds there with no issues.

ok so i was able to rule this out. fedora uses gcc v12 but the arm-none-eabi-gcc is still using v11 which is why compile works on the tests.

> That is weird too (not sure how it thinks casting from unsigned to unsigned can change the sign of anything!); we may be getting to the point where I will have to try it myself, but could you add
> 
> static_assert(sizeof(uint)==4, "uint should be 4 bytes");
> 
> somewhere at the top of rtc.c

as for this. it didnt appear to work im afraid

kilograham, 2022-05-31:
OK, i  will have to play with it in a docker container; i'd just install the official ARM gcc 11 for now

dhalbert, 2022-05-31:
@tannewt has been using the Arch distribution, which keeps very current, and is at gcc12 for arm-none-eabi.

kilograham, 2022-05-31:
> So i was actually able to get the tests to pass by using these changes but i will be honest i have no idea what effects these changes will have on the sdk:

Yeah, that _shouldn't_ be needed; but glad you are unblocked.

fivdi, 2022-05-31:
> OK, i will have to play with it in a docker container; i'd just install the official ARM gcc 11 for now

Assuming the official ARM gcc 11 is gcc11 for arm-none-eabi, how will this help resolving the problem? Isn't it gcc12 for arm-none-eabi that's needed?

kilograham, 2022-05-31:
I meant I will play with GCC 12 in a docker container; I suggested using GCC 11 as a workaround.

kilograham, 2022-06-17:
i can repro this in docker (once i figured out how to use pacman!)

joshuag1000, 2022-06-17:
> i can repro this in docker (once i figured out how to use pacman!)

Good to know that it wasn't my system causing the issue.
(Yeah pacman can be a learning curve)

kilograham, 2022-06-17:
ok, picio-sdk and examples now build with gcc12_fix branch

joshuag1000, 2022-06-17:
> ok, picio-sdk and examples now build with gcc12_fix branch

That's great. I've just run the tests on my computer and they are all now passing as expected. 
Thanks.

joshuag1000, 2022-06-20:
Should i close this issue with #842 being merged into the develop branch?

PR #842 (2022-06-20): Suppress new GCC 12 warning
fixes #840
closed 2022-06-20 by kilograham