raspberrypi/pico-sdk #849: Make error - multiple definitions of '_exit'

Alessa-L, 2022-06-04:
Hi, i'm trying to set up my pico board on a opensuse tumbleweed system with the following packages installed:
- cross-arm-none-gcc11
- cross-arm-none-newlib-devel
- cmake
- gcc
- gcc-c++
- cross-arm-linux-glib-devel
- cross-arm-gcc11

but when i use the "make" command it throws the following error:

/usr/lib64/gcc/arm-none-eabi/11/ld: /usr/lib64/gcc/arm-none-eabi/11/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc.a(lib_a-syscalls.o): in function `_exit':
/home/abuild/rpmbuild/BUILD/newlib-4.1.0/build-regular-dir/arm-none-eabi/thumb/v6-m/nofp/newlib/libc/sys/arm/../../../../../../../../../newlib/libc/sys/arm/syscalls.c:493: multiple definition of `_exit'; CMakeFiles/main.dir/home/elise/Programming/Pico/SDK/src/rp2_common/pico_runtime/runtime.c.obj:runtime.c:(.text._exit+0x0): first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/main.dir/build.make:807: main.elf] Error 1
make[2]: *** [CMakeFiles/Makefile2:1329: CMakeFiles/main.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1336: CMakeFiles/main.dir/rule] Error 2
make: *** [Makefile:124: main] Error 2

Is there anyway to fix this or should i try a VM with raspberry pi OS ?
If this is an inapropriate question for the repo, tell me and i'll delete it. thanks

[error.txt](https://github.com/raspberrypi/pico-sdk/files/8839071/error.txt)

lurch, 2022-06-05:
The instructions in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf work on Debian and Ubuntu, but we've not tested on OpenSUSE. Maybe if you ask on the [forums](https://forums.raspberrypi.com/viewforum.php?f=143) somebody will be able to help you out?

geurtv, 2022-06-05:
Download the official [Arm GNU Toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads) and use that one instead of the one provided by OpenSUSE. It's what I use on Tumbleweed and works just fine.

Alessa-L, 2022-06-06:
> Download the official [Arm GNU Toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads) and use that one instead of the one provided by OpenSUSE. It's what I use on Tumbleweed and works just fine.

Thanks, I will. I think I will ask on opensuse about the newlib package because when I installed it (instead of cross-arm-none-newlib-devel) I receive another error "nosys.specs not found". Probably is a packaging issue

closed 2022-06-06 by Alessa-L

kilograham, 2022-06-06:
yes `nosys.specs` is what it tells it not to include the standard low level system stuff like `_exit`