raspberrypi/pico-sdk #872: fgets(stdin) halts

arnytrty, 2022-06-18:
Using function `fgets` to read one line from stdin halts. Similar to issue https://github.com/raspberrypi/pico-sdk/issues/132.

* Board: RPi Pico
* Code compiled on Windows Machine over WSL (Windows Subsystem for Linux)
* I tried PuTTY on Windows and Minicom on Linux (to try both endline characters)
* USB-UART is something chinese CH340 - never had problem with it before

Sadly, I dont have JTAG so I cant debug. Can somebody please try it to confirm the issue?

### Testing Code
```c
#include <stdio.h>
#include "pico/stdlib.h"

int main(void) {
	stdio_uart_init();

	printf("Hello World! Write anything:\n");

	char buf[128];
	fgets(buf, 128, stdin);

	printf("Output: %c", buf);

	return 0;
}
```

### pico-sdk version
```bash
$ git show -s
commit 426e46126b5a1efaea4544cdb71ab81b61983034 (HEAD -> master, tag: 1.3.1, origin/master, origin/HEAD)
Author: graham sanderson <graham.sanderson@raspberrypi.com>
Date:   Tue May 17 13:42:49 2022 -0500

    SDK 1.3.1 release
```

geurtv, 2022-06-18:
Works as expected here

From https://en.cppreference.com/w/c/io/fgets: Parsing stops if a newline character is found.

Which means keyboard enter (which is carriage return everywhere) doesn't do anything. If from PuTTY I type stuff and then do Alt+Num1Num0 it works just fine.

Note also: you meant to say '%s', not '%c'.

arnytrty, 2022-06-18:
Yeah you are right,

I was thinking I have PuTTY set up correctly to send newline characters, tried sending with alt and it worked.

Thank you.

closed 2022-06-18 by arnytrty