raspberrypi/pico-sdk #895: README doesn't describe how to target Pico W board

lbt, 2022-07-01:
Although the pico_w.h board is now present the 'terse' README should probably mention how to target it

kilograham, 2022-07-01:
which README?

lurch, 2022-07-01:
Presumably this one? https://github.com/raspberrypi/pico-sdk/blob/master/README.md

@lbt There's more detailed documentation in https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf

lbt, 2022-07-02:
Thanks - I knew there would be a target :)

I looked at commit https://github.com/raspberrypi/pico-sdk/commit/5e9a5e827b3265fca53529883abdcc87608abad5 and although the top level README.md is extended, none of the docs mention how to target the newly added board and nor do they add the link mentioned above.

I'd think adding a new board and introducing a new concept of NEEDING to target it would be important enough for the high level README ;)

I'll pull together a patch now @lurch pointed me to the correct syntax

JamesWP, 2022-09-27:
This would have saved me some time too, I've put a comment on the PR to the same effect.

MrGreensWorkshop, 2022-10-11:
I believe adding to CMakeLists.txt file is better than adding as cmake params.
I prefer to put in CMakeLists.txt file like below
`set(PICO_BOARD pico_w)`

kilograham, 2023-01-20:
merged into develop

commit 616e1205 (2022-11-30): Add information on targetting the Pico W. Fixes #895 (#896)

* Add information on targetting the Pico W

Based on comments in https://github.com/raspberrypi/pico-sdk/issues/895 which pointed here:
https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf

I didn't add the pdf link as there are no links to other pdfs.

* Rewording

* stray comma
closed 2023-01-20 by kilograham