raspberrypi/pico-sdk #897: SDK alternatives

Wiz-IO, 2022-07-03:
I've been making an alternative port of Pico SDK for a year and a half who has quite a following and fans
its difference from the official one is expressed in the replacement of CMAKE with SCons ( PlatformIO ) 
and restructuring of the file position in the folders ( I make no changes to the C code )
https://github.com/Wiz-IO/wizio-pico

**Is it possible to post somewhere:
list of all C dependent definitions -D KEY_XXX with a short brief description ( for version )
and compiler verbose logs from 3, 4 basic and important examples**

I need the latter for a quick orientation in building the build because I don't use and don't intend to use the official stuff.
I just have other things on my mind and I don't have a lot of free time, but I do it out of sports malice for those who like it (GIT statistic: ~10 new clones per day ... for 1.5 year )
The port currently works up to version 1.3.1 ... v1.4.0 is in progress - just waiting for the new W board to experiments

I have no direct connections with Pico Team, that's why I'm writing it here. 
Thanks

kilograham, 2022-07-04:
FYI we would (in general) like to provide additional information/tooling to make it easier to build SDK code outside of our build system. I'm obviously not committing on any time frame atm.

Wiz-IO, 2022-07-04:
ОК