raspberrypi/pico-sdk #900: FLASH_BLOCK_SIZE macro conflicts with CYW43 driver

daviesian, 2022-07-03:
The Pico SDK defines a `FLASH_BLOCK_SIZE` macro in [`src/rp2_common/hardware_flash/include/hardware/flash.h`](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_flash/include/hardware/flash.h#L43). The CYW43 driver defines its own `FLASH_BLOCK_SIZE` macro in [`src/cyw43_ll.c`](https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_ll.c#L56). When enabling USB MSC on the Pico W, this conflict prevents compilation. 

A trivial rename of the `FLASH_BLOCK_SIZE` macro in the `pico-sdk` codebase resolves the issue. PR incoming!

daviesian, 2022-07-15:
The CYW43 driver doesn't include `flash.h`, it just happens to define its own macro called `FLASH_BLOCK_SIZE`. So when you enable USB MSC on the Pico W, there are two entirely unrelated parts of the codebase (the CYW43 driver, and the flash driver) which both define a macro `FLASH_BLOCK_SIZE` for their own purposes. This results in a redefinition error, and the macro in the Pico SDK seemed like the easier one to rename :)

```
/Users/ipd21/dev/micropython/lib/cyw43-driver/src/cyw43_ll.c:56: error: "FLASH_BLOCK_SIZE" redefined [-Werror]
   56 | #define FLASH_BLOCK_SIZE (512)
      | 
In file included from /Users/ipd21/dev/micropython/ports/rp2/mpconfigport.h:35,
                 from /Users/ipd21/dev/micropython/py/mpconfig.h:62,
                 from /Users/ipd21/dev/micropython/ports/rp2/cyw43_configport.h:30,
                 from /Users/ipd21/dev/micropython/lib/cyw43-driver/src/cyw43_config.h:38,
                 from /Users/ipd21/dev/micropython/lib/cyw43-driver/src/cyw43_ll.c:40:
/Users/ipd21/dev/micropython/lib/pico-sdk/src/rp2_common/hardware_flash/include/hardware/flash.h:43: note: this is the location of the previous definition
   43 | #define FLASH_BLOCK_SIZE (1u << 16)
```

kilograham, 2022-07-15:
> The CYW43 driver doesn't include flash.h, it just happens to define its own macro called FLASH_BLOCK_SIZE. So when you enable USB MSC on the Pico W, there are two entirely unrelated parts of the [...]

You misunderstand how redefinition errors work. For the error to happen, the cyw43 driver would *have* to have included flash.h. This happens because of this:

https://github.com/micropython/micropython/blob/5bf3765631e645412dbd62a616cfdadeca5ea0c3/ports/rp2/mpconfigport.h#L35

Would have been good to know this was MicroPython to start with; anyways. this would be a backwards incompatible change here, we will need to fix it in either MicroPython or cyw43_driver

peterharperuk, 2022-11-22:
FLASH_BLOCK_SIZE has been renamed to CYW43_FLASH_BLOCK_SIZE, so I don't think there's anything to do here. Lots of "base ports" define FLASH_BLOCK_SIZE, so I think it's reasonable.
Will close if no one disagrees.

daviesian, 2022-11-22:
That's great - if the macro has been renamed in the CYW43 driver then that entirely resolves the conflict. Please feel free to close. Thanks!

peterharperuk, 2022-11-23:
Sorry - forgot that the driver needs to be updated in the sdk!

kilograham, 2022-11-24:
cyw43_driver updated in develop, so closing

PR #1115 (2022-11-24): Update cyw43-driver to latest version
This pulls in a few fixes (7 commits) and catches us up with MP

Fixes #900
commit 6cc9394c (2022-11-24): Update cyw43-driver to latest version (#1115)

Fixes #900
closed 2022-11-24 by kilograham