raspberrypi/pico-sdk #903: Unable to setup build environment on Apple Silicon, failing with: internal compiler error: Illegal instruction: 4

szaydel, 2022-07-04:
I may well be doing something wrong, and hopefully that's it. I followed the guidance in the getting started guide and I think I have things all together, but I continue to see `internal compiler error: Illegal instruction: 4` as in the full output below. Maybe this is a known issue, but I have been unable to find anything to suggest that it is. Any guidance would be great on this.

Thank you!

Host information:
```
Mach kernel version:
	 Darwin Kernel Version 21.3.0: Wed Jan  5 21:37:58 PST 2022; root:xnu-8019.80.24~20/RELEASE_ARM64_T6000
Kernel configured for up to 10 processors.
10 processors are physically available.
10 processors are logically available.
Processor type: arm64e (ARM64E)
Processors active: 0 1 2 3 4 5 6 7 8 9
Primary memory available: 64.00 gigabytes
Default processor set: 611 tasks, 3492 threads, 10 processors
Load average: 1.34, Mach factor: 8.64
```

```
[variant] Loaded new set of variants
[kit] Successfully loaded 3 kits from /Users/szaydel/.local/share/CMakeTools/cmake-tools-kits.json
[main] Building folder: pico-examples 
[proc] Executing command: /opt/homebrew/bin/arm-none-eabi-gcc -v
[main] Configuring folder: pico-examples 
[proc] Executing command: /opt/homebrew/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=/opt/homebrew/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER:FILEPATH=/opt/homebrew/bin/arm-none-eabi-g++ -S/Users/szaydel/pico/pico-examples -B/Users/szaydel/pico/pico-examples/build -G "Unix Makefiles"
[cmake] PICO_SDK_PATH is /Users/szaydel/pico/pico-sdk
[cmake] PICO platform is rp2040.
[cmake] Not searching for unused variables given on the command line.
[cmake] Build type is Debug
[cmake] Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
[cmake] PICO target board is pico.
[cmake] Using board configuration from /Users/szaydel/pico/pico-sdk/src/boards/include/boards/pico.h
[cmake] TinyUSB available at /Users/szaydel/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
[cmake] Compiling TinyUSB with CFG_TUSB_DEBUG=1
[cmake] cyw43-driver available at /Users/szaydel/pico/pico-sdk/lib/cyw43-driver
[cmake] lwIP available at /Users/szaydel/pico/pico-sdk/lib/lwip
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: /Users/szaydel/pico/pico-examples/build
[build] Starting build
[proc] Executing command: /opt/homebrew/bin/cmake --build /Users/szaydel/pico/pico-examples/build --config Debug --target blink -j 12 --
[build] Scanning dependencies of target bs2_default
[build] [  0%] Performing build step for 'ELF2UF2Build'
[build] [  0%] Built target bs2_default
[build] [  0%] Built target bs2_default_padded_checksummed_asm
[build] [100%] Built target elf2uf2
[build] [  0%] No install step for 'ELF2UF2Build'
[build] [  0%] Completed 'ELF2UF2Build'
[build] [  0%] Built target ELF2UF2Build
[build] Scanning dependencies of target blink
[build] Consolidate compiler generated dependencies of target blink
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_claim/claim.c.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_sync/sync.c.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/pico_printf/printf.c.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/common/pico_util/pheap.c.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_clocks/clocks.c.obj
[build] [  0%] Building ASM object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/pico_bit_ops/bit_ops_aeabi.S.obj
[build] [  0%] Building C object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/pico_bootrom/bootrom.c.obj
[build] [  0%] Building ASM object blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/pico_divider/divider.S.obj
[build] /Users/szaydel/pico/pico-sdk/src/rp2_common/pico_printf/printf.c: In function '_ftoa':
[build] /Users/szaydel/pico/pico-sdk/src/rp2_common/pico_printf/printf.c:354:5: internal compiler error: Illegal instruction: 4
[build]   354 |     if (value < -DBL_MAX)
[build]       |     ^~
[build] during GIMPLE pass: profile_estimate
[build] /Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_claim/claim.c: In function 'hw_claim_unused_from_range':
[build] /Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_claim/claim.c:54:1: internal compiler error: Illegal instruction: 4
[... 63 lines cut ...]
[build] gmake[3]: *** [blink/CMakeFiles/blink.dir/build.make:146: blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_sync/sync.c.obj] Error 1
[build] gmake[3]: *** [blink/CMakeFiles/blink.dir/build.make:90: blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj] Error 1
[build] during GIMPLE pass: profile_estimate
[build] /Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c: In function 'gpio_default_irq_handler':
[build] /Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c:263:1: internal compiler error: Illegal instruction: 4
[build]   263 | }
[build]       | ^
[build] Please submit a full bug report,
[build] with preprocessed source if appropriate.
[build] See <https://bugs.linaro.org/> for instructions.
[build] during GIMPLE pass: profile_estimate
[build] /Users/szaydel/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c: In function 'runtime_init':
[build] /Users/szaydel/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c:289:1: internal compiler error: Illegal instruction: 4
[build]   289 | }
[build]       | ^
[build] Please submit a full bug report,
[build] with preprocessed source if appropriate.
[build] See <https://bugs.linaro.org/> for instructions.
[build] gmake[3]: *** [blink/CMakeFiles/blink.dir/build.make:104: blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj] Error 1
[build] gmake[3]: *** [blink/CMakeFiles/blink.dir/build.make:354: blink/CMakeFiles/blink.dir/Users/szaydel/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c.obj] Error 1
[build] gmake[2]: *** [CMakeFiles/Makefile2:4374: blink/CMakeFiles/blink.dir/all] Error 2
[build] gmake[1]: *** [CMakeFiles/Makefile2:4381: blink/CMakeFiles/blink.dir/rule] Error 2
[build] gmake: *** [Makefile:202: blink] Error 2
[build] Build finished with exit code 2
```

lurch, 2022-07-04:
Have you installed Rosetta? See Chapter 9 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

szaydel, 2022-07-04:
Thank you for such a quick response Andrew, I have, but I am not sure how to confirm that it is working as expected. I have been running some apps which were built for x86, but I am wondering if there is some way I can validate whether that's in fact part of the problem here.

lurch, 2022-07-04:
> I am not sure how to confirm that it is working as expected

Me neither, I don't have an Apple Silicon Mac.

szaydel, 2022-07-04:
Thank you, my only confirmation has been through using the apps which run on my intel mac, directly without doing anything to them on the M1 macbook. I am fairly confident that if something was wrong those apps would not launch. Though maybe something about the build environment is preventing proper interaction.

lurch, 2022-07-04:
> internal compiler error: Illegal instruction

It might be worth uninstalling and reinstalling your compiler? And checking that you've actually installed the right compiler (and version)? :man_shrugging:

szaydel, 2022-07-04:
So, this is where things get interesting. :) I actually noticed that I had a `/Applications/ARM` directory, which I was unsure about, but thought, let me move it out of the way. I did that, and then with homebrew removed and reinstalled the compiler. It looks like things are actually working fine now, which makes me wonder if this directory was contributing to the environment without me recognizing.

kilograham, 2022-07-04:
Be worth checking which version of GCC you are now running. My personal experience is that GCC 11 (the latest version available from ARM) does not work on M1.. previous versions do.

DatanoiseTV, 2022-07-05:
@szaydel 

Could you please try
```
brew install cmake
brew tap ArmMbed/homebrew-formulae
brew install arm-none-eabi-gcc
``

szaydel, 2022-07-05:
About GCC version:
```
$ brew info arm-none-eabi-gcc
armmbed/formulae/arm-none-eabi-gcc: stable 10.3-2021.07
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
/opt/homebrew/Cellar/arm-none-eabi-gcc/10.3-2021.07 (6,790 files, 715MB) *
  Built from source on 2022-07-04 at 08:17:46
From: https://github.com/ArmMbed/homebrew-formulae/blob/HEAD/arm-none-eabi-gcc.rb
```

@DatanoiseTV, that is exactly what I did yesterday, and then after discovering `/Applications/ARM` and moving it out of the way, I re-installed `arm-none-eabi-gcc`, at which point things appeared to be working.

lurch, 2022-07-14:
@szaydel What was _in_ your `/Applications/ARM` directory?

jsphpl, 2022-07-14:
> I am actually still unsure where the /Applications/ARM directory came from

https://github.com/ArmMbed/homebrew-formulae/blob/HEAD/arm-none-eabi-gcc.rb downloads and runs the installer from developer.arm.com, which creates the /Applications/ARM directory.

szaydel, 2022-07-14:
Ahhh, OK that makes sense. @lurch , I can list recursively contents of that directory, but not sure it will be all that helpful. Here's top level:
```
11.2-2022.02-darwin-x86_64-arm-none-eabi-manifest.txt  lib/
arm-none-eabi/                                         libexec/
bin/                                                   share/
include/
```

And, I want to say I am using `gcc` from a different directory, after re-installing with `homebrew`:
```
$ which arm-none-eabi-gcc
/opt/homebrew/bin/arm-none-eabi-gcc

$ ls /Applications/zzz.ARM/bin/arm-none-eabi-gcc
/Applications/zzz.ARM/bin/arm-none-eabi-gcc*
```
So from what I can tell, I must have been using the later 11.2 version from `/Applications/ARM` until I renamed it and reinstalled with `homebrew`.

szaydel, 2022-07-14:
That's my barely informed guess as well. I am just not sure how I ended up having two different versions, but maybe I will unpack that mystery at some point. And yes, let's close this out. Thanks for all the help folks!

szaydel, 2022-07-14:
Thanks for the tip on the HB issue @jsphpl, I appreciate it.