raspberrypi/pico-sdk #906: CDC doesn't respect no flow control setting (DTR control needed to receive data)

MrGreensWorkshop, 2022-07-08:
### Operating System

Windows, Mac OS, I think in all OS es.

### Board

Raspberry pi pico

### Firmware

just simple app to test cdc behavior.

https://github.com/MrGreensWorkshop/RasPiPicoSDK_cdc_test

- Download
- Compile and run. (or just use binary named PicoTest.uf2)
- Use Terminal.exe or any other terminal you like to check DTR behavior. 

### What happened ?

When handshake is set to none, to receive data we need to enable DTR. We shouldn't be force to enable DTR to receive data. (other flow control status shouldn't be changed in cdc side if it is the case.)

I believe this caused by tinyusb lib. That's why I open this issue also in [tinyusb repo](https://github.com/hathach/tinyusb/issues/1548).

please watch the gif file.

[image]

### How to reproduce ?

https://github.com/MrGreensWorkshop/RasPiPicoSDK_cdc_test

- Download
- Compile and run. (or just use binary named PicoTest.uf2)
- Use Terminal.exe or any other terminal you like to check DTR behavior as shown in gif file.

### Screenshots

This the demonstration that 2 FDTI USB serial converter connected each other with 3 pins.
```
Rx <- Tx
Tx -> Rx
GND <-> GND
```
It shouldn't be forced to enable DTR to receive data either sides. Please watch the gif file.

[image]

lurch, 2022-07-08:
Are you sure that this isn't a quirk of your particular serial terminal program?  I never had to fiddle with any DTR settings when using PuTTY as detailed in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

MrGreensWorkshop, 2022-07-08:
PuTTY or Teraterm terminals are made for VT100, not plain text transmission. You can use if you want. But those are uses other pins too. DTR -> DSR  and DSR <-  DTR. That's the problem. Asking host application for DSR on is reduce the compatibility with other apps. I'm saying this because if you check 2 FDTI USB serial converter example gif, it doesn't uses DTR. Most 3rd party apps only uses RX, TX, GND if no flow control involved.

aallan, 2022-07-08:
I've never used any flow control when talking to an RP2040-based board. I've never even connected up the CTS/RTS or DSR/DTR pins when using UART adaptors, just RX/TX and GND. I can't replicate this problem. Downloading the UF2 binary in the OP's repo it works as expected on my machine.

[image]

[image]

closed 2022-07-08 by aallan

MrGreensWorkshop, 2022-07-08:
you used screen command right? its also uses VT100

aallan, 2022-07-08:
> you used screen command right? its also uses VT100

I did not use the `screen` command.

[image]

MrGreensWorkshop, 2022-07-08:
This is my setup. 

Raspberry pi pico -> USB thats all. No usb serial converters attached. 

This happens when raspberry pi usb cdc is used.

This is other issue talks about same issue. 
https://github.com/hathach/tinyusb/discussions/932

Please check this file.

https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c

it is checking DTR if it's on or not.
```
bool tud_cdc_n_connected(uint8_t itf)
{
  // DTR (bit 0) active  is considered as connected
  return tud_ready() && tu_bit_test(_cdcd_itf[itf].line_state, 0);
}
```

aallan, 2022-07-08:
The UF2 binary you provided doesn't have CDC enabled, only UART?

MrGreensWorkshop, 2022-07-08:
Yes, I checked again just to be 100% sure. I downloaded the file and write to my pico. Yes I only check CDC over USB. I didn't even check uart output because you can clearly see my setting that  I disabled UART I'm not talking about uart.

https://github.com/MrGreensWorkshop/RasPiPicoSDK_cdc_test/blob/main/CMakeLists.txt

```
# Enable USB CDC
pico_enable_stdio_usb(PicoTest 1)

# Disable UART
pico_enable_stdio_uart(PicoTest 0)
 
# Enable extra outputs (bins)
pico_add_extra_outputs(PicoTest)
```

MrGreensWorkshop, 2022-07-08:
I mentioned [erlier here](https://github.com/raspberrypi/pico-sdk/issues/906#issuecomment-1178758803) if you use VT100 terminal you cant catch it.

My I kindly ask you, please check with [Coolterm](https://freeware.the-meiers.org/CoolTermMac.zip) as I mentioned [here](https://github.com/raspberrypi/pico-sdk/issues/906#issuecomment-1178700704) 

Thank you.

MrGreensWorkshop, 2022-07-08:
This happens when raspberry pi usb cdc is used.

This is other issue talks about same thing.
https://github.com/hathach/tinyusb/discussions/932

Please check this file.

https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c

it is checking DTR if it's on or not regardless checking flow control.

bool tud_cdc_n_connected(uint8_t itf)
{
  // DTR (bit 0) active  is considered as connected
  return tud_ready() && tu_bit_test(_cdcd_itf[itf].line_state, 0);
}

JamesH65, 2022-07-08:
Might be worth flagging this up on the tinyUSB GitHub issue tracker, since it is their code.

MrGreensWorkshop, 2022-07-09:
"can't reproduce" feels sad.

MrGreensWorkshop, 2022-07-19:
@JamesH65
> Might be worth flagging this up on the tinyUSB GitHub issue tracker, since it is their code.

it turns out the problem is how tinyUSB implemented in pico-sdk source code.
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/stdio_usb.c

`tud_cdc_connected` is used for checking connection but under the hood its checking DTR. I don't think using `tud_cdc_connected` is appropriate way to check the connection.

Could you please consider to reopen the issue please?

[src/class/cdc/cdc_device.h](https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.h)
```
...
static inline bool tud_cdc_connected (void)
{
  return tud_cdc_n_connected(0);
}
...
```

[src/class/cdc/cdc_device.h](https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c)

```
...
bool tud_cdc_n_connected(uint8_t itf)
{
  // DTR (bit 0) active  is considered as connected
  return tud_ready() && tu_bit_test(_cdcd_itf[itf].line_state, 0);
}
...
```

JamesH65, 2022-07-19:
Not my call to reopen, @kilograham?