raspberrypi/pico-sdk #912: Long connect time with Zyxel AP when power saving is on

maxnet, 2022-07-09:
Even when the mesh setting is disabled, I am continuing to experience that it often (but not always) takes unacceptable long to join the network when using a Zyxel VMG8825-T50 access point.
Which is a pretty common router in Europe, as T-mobile (which has millions of customers) provides it to their clients.

Attached wifi capture .pcapng files (done on a laptop with Realtek chipset and wifi set to monitoring mode) of when it connects reasonable well in 4 seconds, and when it takes 10 seconds.
Using latest daily micropython build (rp2-pico-w-20220708-unstable-v1.19.1-113-g9af6a275d), but as I believe the problem is not limited to micropython, I am creating an issue here.

[image]

When it takes a long time, you can see that the AP sends an "Association response" like it should, but it seems like the Pico is not taking notice of it, as the Pico does nothing for 3 seconds, and then restarts the joining process from the start with "Probe requests".
And also the same problem of the Pico not responding to packets happens a second time with a reassociation response .

When I disable power saving with "wlan.config(pm = 0xa11140)" it does seem to connect reliably in 3-4 seconds.
So perhaps make that the default?

[pico-wifi-capture.zip](https://github.com/raspberrypi/pico-sdk/files/9076369/pico-wifi-capture.zip)

kilograham, 2022-07-09:
cc @peterharperuk

maxnet, 2022-07-11:
42 NOT good. 7 seconds to connect.

peterharperuk, 2022-07-11:
Can you try one more? 0x111022

maxnet, 2022-07-11:
>Can you try one more? 0x111022

0x111022 not good either.
"Got wifi, took 7 seconds"

Not sure what you are trying to do though, wouldn't that make it to go to power saving even faster, reducing time before it goes to sleep from 200 ms (second last digits 14 hex), to 20 ms (second last digits 02) ?

https://github.com/georgerobotics/cyw43-driver/blob/5614e2750e019b0ffa14a50e3c92045b952e2634/src/cyw43.h#L523

peterharperuk, 2022-07-11:
Yes - I think you might be right. I'm not sure we've got CYW43_AGGRESSIVE_PM and CYW43_PERFORMANCE_PM the right way around.

Do you see any difference when using 0xa11c82?

maxnet, 2022-07-11:
>Do you see any difference when using 0xa11c82?

Hard to tell if it is any better.
Seeing 6 and 7 seconds connection time, mixed with the good 4 seconds (which you do get reliably without power saving, with pm values ending on 0).

Wondering if power saving can be disabled just during authentication, as I think during that phase messages are not resend when you miss one, while that do may be the case with normal traffic...

peterharperuk, 2022-07-11:
Yes. I might send you something to test if that's ok?

peterharperuk, 2023-02-13:
This got stuck waiting for this change https://github.com/georgerobotics/cyw43-driver/pull/10
We could probably fix it in the sdk by disabling power management until a connection succeeds?

kilograham, 2023-05-26:
what is the current state of this?

kilograham, 2023-05-27:
ha, i guess i should look at my labels!

peterharperuk, 2024-05-21:
Not an sdk issue. See https://github.com/georgerobotics/cyw43-driver/pull/10

closed 2024-05-21 by peterharperuk