raspberrypi/pico-sdk #915: Pico W strange issue sending but not receiving network packets

pgreenland, 2022-07-11:
Hi,

I've just started evaluating the Pico W for use in a future project but have been seeing some very strange things when trying to use the Wi-Fi.

Built the SDK examples without any issues and flashed using another Pico as a programmer / debugger, all going well.

First test with the iperf freertos server example, flashing "picow_freertos_iperf_server_sys.elf" went well. I was able to connect and achieve around 25Mbits/s.

Started working on my own application based on FreeRTOS and a stripped down version of the iPerf server application.

Started to experience the occasional Wi-Fi connection issue, often the first connection after a long period of activity would go well. The pi is able to receive traffic, responding to pings etc.

After a few programming / reboot cycles things start to go wrong, for example DHCP completes but no further traffic is received by the Pi.

It seems as if the module isn't able to receive Ethernet packets anymore, but is able to send them.

I reverted back to the pico examples to try to work out what's going on, using a debug build of the iperf example built with:

```
cd pico-examples
mkdir build
cd build
export PICO_TOOLCHAIN_PATH=/Applications/ARM/bin
export PICO_SDK_PATH=/Users/phil/Repos/pico/pico-sdk
export PICO_BOARD=pico_w
export FREERTOS_KERNEL_PATH=/Users/phil/Repos/pico/FreeRTOS-Kernel
cmake -DWIFI_SSID="Redacted" -DWIFI_PASSWORD="Redacted" -DCMAKE_BUILD_TYPE=Debug ..
cd pico_w/freertos/iperf
make -j8
```

When attempting connection to the Wi-Fi network the following is seen on the serial console:

```
Starting FreeRTOS on core 0:
Version: 7.95.49 (2271bb6 CY) CRC: b7a28ef3 Date: Mon 2021-11-29 22:50:27 PST Ucode Ver: 1043.2162 FWID 01-c51d9400
cyw43 loaded ok, mac 28:cd:c1:00:b6:e9
API: 12.2
Data: RaspberryPi.PicoW
Compiler: 1.29.4
ClmImport: 1.47.1
Customization: v5 22/06/24
Creation: 2022-06-24 06:55:08
Connecting to WiFi...
connect status: joining
connect status: no ip
failed to connect.
```

On the access point running OpenWRT, a packet capture on the Wi-Fi interface shows both DHCP requests and responses from the DHCP server, which appear to not reach the Pi

```
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
22:33:53.349795 28:cd:c1:00:b6:e9 > ff:ff:ff:ff:ff:ff Null Unnumbered, xid, Flags [Response], length 6: 01 00
22:33:53.349928 28:cd:c1:00:b6:e9 > ff:ff:ff:ff:ff:ff Null Unnumbered, xid, Flags [Response], length 6: 01 00
22:36:29.239763 28:cd:c1:00:b6:e9 > ff:ff:ff:ff:ff:ff Null Unnumbered, xid, Flags [Response], length 6: 01 00
22:36:29.239898 28:cd:c1:00:b6:e9 > ff:ff:ff:ff:ff:ff Null Unnumbered, xid, Flags [Response], length 6: 01 00
22:36:30.865037 IP (tos 0x0, ttl 255, id 1, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from 28:cd:c1:00:b6:e9, length 308, xid 0x6054778a, Flags [none] (0x0000)
	  Client-Ethernet-Address 28:cd:c1:00:b6:e9
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    MSZ Option 57, length 2: 1500
	    Parameter-Request Option 55, length 4: 
	      Subnet-Mask, Default-Gateway, BR, Domain-Name-Server
22:36:30.865116 IP (tos 0x0, ttl 255, id 1, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from 28:cd:c1:00:b6:e9, length 308, xid 0x6054778a, Flags [none] (0x0000)
	  Client-Ethernet-Address 28:cd:c1:00:b6:e9
	  Vendor-rfc1048 Extensions
[... 172 lines cut ...]
     irq_set_enabled(IO_IRQ_BANK0, true);
 
     cyw43_task_should_exit = false;
-    xTaskCreate(cyw43_task, "CYW43 Worker", configMINIMAL_STACK_SIZE, NULL, CYW43_TASK_PRIORITY, &cyw43_task_handle);
+    xTaskCreate(cyw43_task, "CYW43 Worker", configMINIMAL_STACK_SIZE * 10, NULL, CYW43_TASK_PRIORITY, &cyw43_task_handle);
 #if configUSE_CORE_AFFINITY && configNUM_CORES > 1
     // the cyw43 task mus tbe on the same core so it can restore IRQs
     vTaskCoreAffinitySet(cyw43_task_handle, 1 << portGET_CORE_ID());
@@ -248,4 +251,4 @@ void cyw43_delay_us(uint32_t us) {
 void cyw43_arch_poll() {
 }
 
-#endif
\ No newline at end of file
+#endif
```
Ignore the stack size increase, which was added to resolve a stack overflow when enabling as much debug as I could.

Have tried both with and without the patch, no change in failed case.

Any pointers would / things to check would be appreciated.

Thanks,

Phil

kilograham, 2022-07-12:
you should not need to add `sys_check_timeouts` as lwIP in NO_SYS=0 mode has its own timer/task... (maybe not everything is configured correctly tho!?)

have you tried rebooting your AP?

pgreenland, 2022-07-12:
@kilograham Thanks for the fast reply.

Checked again this morning, same problem, a tactical reboot of the AP when nobody was around solved it immediately.

Strange that I've seen this with two separate APs now though. The one I've most recently been having issues with contains a "MediaTek MT7622 802.11bgn", which seems generally well behaved until it met the Pico :-P.

Seems it still requires my little tweak though, here's a boot of the iperf demo without it:

```
.Starting FreeRTOS on core 0:
dhcp_release_and_stop()
Version: 7.95.49 (2271bb6 CY) CRC: b7a28ef3 Date: Mon 2021-11-29 22:50:27 PST Ucode Ver: 1043.2162 FWID 01-c51d9400
cyw43 loaded ok, mac 28:cd:c1:00:b6:e9
API: 12.2
Data: RaspberryPi.PicoW
Compiler: 1.29.4
ClmImport: 1.47.1
Customization: v5 22/06/24
Creation: 2022-06-24 06:55:08
dhcp_release_and_stop()
dhcp_start(netif=2000102C) w00
dhcp_start(): restarting DHCP configuration
dhcp_start(): starting DHCP configuration
Connecting to WiFi...
connect status: joining
dhcp_discover()
transaction id xid(30958aea)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 2000 msecs
connect status: no ip
failed to connect.
```

Looks like the DHCP discover is a single shot, stop and declare failure.

Here's the output with my hacky patch reapplied:

```
Starting FreeRTOS on core 0:
dhcp_release_and_stop()
Version: 7.95.49 (2271bb6 CY) CRC: b7a28ef3 Date: Mon 2021-11-29 22:50:27 PST Ucode Ver: 1043.2162 FWID 01-c51d9400
cyw43 loaded ok, mac 28:cd:c1:00:b6:e9
API: 12.2
Data: RaspberryPi.PicoW
Compiler: 1.29.4
ClmImport: 1.47.1
Customization: v5 22/06/24
Creation: 2022-06-24 06:55:08
dhcp_release_and_stop()
dhcp_start(netif=2000102C) w00
dhcp_start(): restarting DHCP configuration
dhcp_start(): starting DHCP configuration
Connecting to WiFi...
connect status: joining
dhcp_discover()
transaction id xid(840a7388)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 2000 msecs
connect status: no ip
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
transaction id xid(840a7388)
[... 136 lines cut ...]
dhcp_select: REQUESTING
dhcp_select(): set request timeout 2000 msecs
dhcp_recv(pbuf = 2000A068) from DHCP server 192.168.1.254 port 67
pbuf->len = 300
pbuf->tot_len = 300
skipping option 28 in options
searching DHCP_OPTION_MESSAGE_TYPE
DHCP_ACK received
dhcp_bind(netif=2000102C) w00
dhcp_bind(): t0 renewal timer 86400 secs
dhcp_bind(): set request timeout 86400000 msecs
dhcp_bind(): t1 renewal timer 43200 secs
dhcp_bind(): set request timeout 43200000 msecs
dhcp_bind(): t2 rebind timer 75600 secs
dhcp_bind(): set request timeout 75600000 msecs
dhcp_bind(): IP: 0xe201a8c0 SN: 0x00ffffff GW: 0xfe01a8c0
connect status: link up
Connected.

Ready, running iperf server at 192.168.1.226
locked tcpip core <- lock acquired by tcpip thread
tcpip_timeouts_mbox_fetch sleeptime = 267
```

Seems that the lock holding up the timer processing is only acquired after the connection has been completed....doh.

pgreenland, 2022-07-12:
@peterharperuk Gave that power management option a try, adding it immediately after the cyw43_arch_enable_sta_mode(); call in the freertos iperf demo. No change in getting it connected but it has got the ping times down from the usual 200 to 300ms with occasional timeout down to a steady 2ms. I approve this change :-).

It seems the pico goes through three stages of working with this AP. After an AP reboot it's fine for some number of connections. Then for a period it completes DHCP receiving an IP, before stopping responding to packets. It'll do this several times, until finally it fails to complete DHCP at all until the AP is rebooted again. Each time I can can see requests arriving from the pico on the AP's wireless interface, and the replies from the DHCP server supposedly going out on the wifi interface.

kilograham, 2022-07-12:
ah yes; `WFE` should not be being used here on FreeRTOS :-( ... I guess I was mostly testing with SMP which likely avoids the starvation problem... probably worth adding a separate issue for the WFE thing

kilograham, 2023-05-26:
closing this as #917 was fixed in 1.5.0

closed 2023-05-26 by kilograham