raspberrypi/pico-sdk #917: Pico W cyw43_arch_wifi_connect_timeout_ms under FreeRTOS blocks DHCP retries

pgreenland, 2022-07-12:
When connecting to Wi-FI using cyw43_arch_wifi_connect_timeout_ms under FreeRTOS as per most the examples, DHCP retries are not sent. Therefore if the DHCP server contacted doesn't respond to the first request the connection will fail.

This appears to be caused by the use of best_effort_wfe_or_timeout within cyw43_arch_wifi_connect_until, which is called by cyw43_arch_wifi_connect_timeout_ms.

The use of wfe / sve here combined with the task priorities of the main and tcpip task in the demo project, leads to the tcpip task being starved of processor time and not able to transmit DHCP discover / request retries.

Increasing the priority of the tcpip task over that of the main task via adding the following to lwipopts.h:

```
#define TCPIP_THREAD_PRIO (3UL)
```

Resolves the issue.

Alternatively calling cyw43_arch_wifi_connect_async and followed by periodically monitoring the status of the link via cyw43_tcpip_link_status with a delay also works.

kilograham, 2023-01-18:
Can you please retest with #1177

peterharperuk, 2023-01-18:
It works for me

kilograham, 2023-01-24:
#1177 merged into develop

closed 2023-01-24 by kilograham