raspberrypi/pico-sdk #919: cyw43_arch_poll unneccessary IRQ's

essele, 2022-07-14:
Hi,

The current implementation of cyw43_arch_poll includes a GPIO IRQ mechanism presumably for triggering activity. The trigger for this is currently ignored (commented out) and the poll is executed every time anyway, however the IRQ setup and the ISR is left in place to unnecessarily consume resource.

Can this be tidied? And I'm with whoever left the note for Graham ... there's a polling function which is called regularly, do we really need to complicate things with an IRQ in this model ... surely you could just check the GPIO interrupt status in the polling function and avoid the unnecessary ISR?

Lee.

kilograham, 2023-01-18:
this issue is somewhat obsoleted by #1177

kilograham, 2023-01-20:
please re-open if you feel strongly... pico_cyw43_driver is now agnostic of the polling mode, however you could add a new pico_cyw43_driver variant that didn't use IRQs.

closed 2023-01-20 by kilograham