raspberrypi/pico-sdk #921: stdio over USB needs DTR signal to be activated from terminal program

RobertK66, 2022-07-14:
I think, this is no bug but a feature ;-).
But at least, IMHO it would be worth a note in the documentation. (Chapter 4.4 in 'Getting started ...')

I wasted some time in debugging my Windows 11 - USB driver Stack and tried to understand your documented issues with the 'USB Enumeration workaround' .....

In the end, I found that using the HTerm terminal program is my issue! This terminal software does not set DTR signal on connect per default! Also my own code using NET Serial Port needed the line:

`  Port.DtrEnable = true;`

to be added in order to work. Using putty iso HTerm worked out of the box.
( But once you know the Problem, there is a Button fot DTR in HTerm 😄 ... )

lurch, 2022-07-14:
This was also discussed recently in #906 - I've not been following it very closely, but it seems to be more of a tinyusb thing than a pico-sdk thing? https://github.com/hathach/tinyusb/issues/1548 :shrug:

RobertK66, 2022-07-14:
Ok, Thx for pointing me there.
I am not sure, if "DTR" is considered part of 'Flow Control' but I also didn't expect the behaviour it currently shows....

RobertK66, 2022-07-14:
Note (to others having problems to get this up and running): 
I still need the 
`TUD_OPT_RP2040_USB_DEVICE_ENUMERATION_FIX=1`
to make the com port working on my Windows 11 Notebook. If I leave this to default or 0, then the COM port shows up in device manager but is 'not connectable' (Hterm hangs on connect).

lurch, 2022-07-14:
Whether you need the USB_DEVICE_ENUMERATION_FIX or not depends which other USB devices you have connected at the same time. And it's also not needed if you're using a board using RP2040 B2, e.g. Raspberry Pi Pico W.

kilograham, 2022-08-10:
merged into develop

closed 2022-08-10 by kilograham