raspberrypi/pico-sdk #923: dma_channel_abort does not wait for CHAN_ABORT to go zero

alastairpatrick, 2022-07-17:
The source for `dma_channel_abort` from SDK 1.4.0:
```c
static inline void dma_channel_abort(uint channel) {
    check_dma_channel_param(channel);
    dma_hw->abort = 1u << channel;
    // Bit will go 0 once channel has reached safe state
    // (i.e. any in-flight transfers have retired)
    while (dma_hw->ch[channel].ctrl_trig & DMA_CH0_CTRL_TRIG_BUSY_BITS) tight_loop_contents();
}
```

Note that while it waits for the channel's `BUSY` flag to go zero, it does _not_ wait for `CHAN_ABORT` to go zero.

From the datasheet, emphasis mine:
> DMA: CHAN_ABORT Register
> Offset: 0x444
> Description
> Abort an in-progress transfer sequence on one or more channels
> 
> Each bit corresponds to a channel. Writing a 1 aborts whatever transfer sequence is in progress on that channel. The bit will remain high until any inflight transfers have been flushed through the address and data FIFOs.
> 
> **After writing, this register must be polled until it returns all-zero. Until this point, it is unsafe to restart the channel.**

Also from the datasheet:
> At the time an abort is triggered, a channel may have bus transfers currently in flight between the read and write master,
> and these transfers cannot be revoked. The CTRL.BUSY flag stays high until these transfers complete, and the channel
> reaches a safe state, which generally takes only a few cycles. **The channel must not be restarted until its CTRL.BUSY flag
> deasserts.** Starting a new sequence of transfers whilst transfers from an old sequence are still in flight can lead to
> unpredictable behaviour.

The datasheet does not indicate whether it is necessary to wait for both `BUSY` to go zero _and_ `CHAN_ABORT` to go zero or if waiting for either condition is sufficient. It also does not indicate in which order code should wait for them to go zero. As mentioned, dma_channel_abort only waits for `BUSY` to go zero.

I suggest there is either a bug in `dma_channel_abort` (if it ought to be waiting for both) or the datasheet would benefit from clarification.

lurch, 2022-07-17:
See also Errata RP2040-E13 in the [RP2040 datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf), if you haven't already.

alastairpatrick, 2022-07-17:
Given RP2040-E13, I agree that this is documentation.

The workaround recommends "Before aborting a channel, clear its interrupt enable. After aborting a channel, poll the CTRL.BUSY bit to wait for completion (not the ABORT bit), clear the spurious IRQ, and restore the interrupt enable."

I suggest this could be improved too. CHAN_ABORT can be used to abort _multiple_ channels with a single write to the register. In fact this seems like the best way to abort multiple channels that trigger each other, with other methods likely to suffer from race conditions.

The workaround recommendation assumes that only a _single_ channel is being aborted. It would be good to confirm that, even when _multiple_ channels are aborted, it is adequate and necessary to check only the CTRL.BUSY bits for each channel in turn and not to check CHAN_ABORT. Without that clarification, it is not clear that checking only CTRL.BUSY is free of race conditions when aborting multiple channels.

kilograham, 2023-05-26:
moving to 1.5.1 so it doesn't get forgotten (even if the docs aren't fixed there)

tommie, 2024-04-29:
> In fact this seems like the best way to abort multiple channels that trigger each other, with other methods likely to suffer from race conditions.

Related, and for posterity, I have a DMA control+data setup, with the data channel re-starting control at the end, rather than issuing a null transfer, emulating a ring. This required me to do two CHAN_ABORT on both channels to ensure it stops. I'm thinking the DMA trigger register may already be in the DMA FIFO, making this racy even with CHAN_ABORT. To be safe, I have it in a loop:

```c
while (dma_channel_is_busy(dma_ctrl) || dma_channel_is_busy(dma_data)) {
    dma_channel_abort_mask((1u << dma_ctrl) | (1u << dma_data));
}
```