raspberrypi/pico-sdk #924: Include directories should be added as system include directories

ooxi, 2022-07-17:
Currently all include directories are added as none system include directories. This causes warnings and errors when compiling my own source code

```/pico-sdk/src/rp2_common/hardware_gpio/include/hardware/gpio.h:882:33: error: ISO C++ does not permit named variadic macros [-Werror=variadic-macros]
  882 | #define CU_REGISTER_DEBUG_PINS(p...) \
      |                                 ^~~
In file included from /pico-sdk/src/common/pico_base/include/pico/types.h:12,
                 from /pico-sdk/src/common/pico_base/include/pico.h:22,
                 from /pico-sdk/src/common/pico_stdlib/include/pico/stdlib.h:10,
                 from /blink.c++:2:
```

ooxi, 2022-07-20:
@kilograham I'm using Fedora 36, this is the compiler identification according to CMake

```
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
```

I have created a minimal example on GitHub where you can see the full compiler output https://github.com/ooxi/pico-sdk-pedantic/runs/7435146674?check_suite_focus=true

kilograham, 2022-07-20:
The SDK will not compile with both `-Wpedantic` and '-Werror`

ooxi, 2022-07-20:
@kilograham first of all, thank you very much for your fast reaction! Very appreciated :-)

> The SDK will not compile with both `-Wpedantic` and `-Werror`

I do not want to compile the SDK with `-Wpedantic` and `-Werror`, just my own sources. Therefore I used `set_source_files_properties` to set those compile options only on my own sources, not all sources of the library.

Using `target_include_directories` with `SYSTEM INTERFACE` instead of `INTERFACE` will let the compiler know to not apply those settings on the include.

I will come back with a proof of concept :-)

ooxi, 2022-07-20:
@kilograham I have created a [proof of concept](https://github.com/ooxi/pico-sdk/commit/afb46fe8d6c76347e5bd2f09b06dca363ab21bee) which is sufficient to [compile the example with `-Wpedantic and -Werror`](https://github.com/ooxi/pico-sdk-pedantic/runs/7436328429?check_suite_focus=true)

I would be happy to clean up the patch and prepare a PR if you consider accepting it.

kilograham, 2022-07-20:
Ah, ok i see the point... warnings are suppressed in system headers.

ooxi, 2022-07-21:
@kilograham feel free to assign the issue to me, I will prepare a PR against develop

kilograham, 2022-12-06:
duplicate of #82

closed 2024-08-08 by kilograham