raspberrypi/pico-sdk #925: Allow the changes of "USBD_STR_PRODUCT"

RaresAil, 2022-07-17:
Is possible to allow the product name change for the board without changing the SDK locally? Or to fork it and make the changes there? I seen that the current value "Pico" is hardcoded there.

lurch, 2022-07-17:
You might want to also have a look at https://github.com/raspberrypi/usb-pid if you're creating your own device based around the RP2040?

RaresAil, 2022-07-17:
I am not selling the device, i want a way to identify that the specific Pico board is running the current firmware

MrGreensWorkshop, 2022-07-20:
Hi, Look how USBD_STR_SERIAL is implemented in [stdio_usb_descriptors.c](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/stdio_usb_descriptors.c) it is using unique flash id. So it is always unique, you can use usb device serial no.

```
static char usbd_serial_str[PICO_UNIQUE_BOARD_ID_SIZE_BYTES * 2 + 1];

static const char *const usbd_desc_str[] = {
    [USBD_STR_MANUF] = "Raspberry Pi",
    [USBD_STR_PRODUCT] = "Pico",
    [USBD_STR_SERIAL] = usbd_serial_str,
    [USBD_STR_CDC] = "Board CDC",
#if PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE
    [USBD_STR_RPI_RESET] = "Reset",
#endif
};
...
const uint16_t *tud_descriptor_string_cb(uint8_t index, __unused uint16_t langid) {
...
    // Assign the SN using the unique flash id
    if (!usbd_serial_str[0]) {
        pico_get_unique_board_id_string(usbd_serial_str, sizeof(usbd_serial_str));
    }
...
}
```

closed 2022-07-23 by RaresAil