raspberrypi/pico-sdk #927: Question regarding SDK 1.3.1 and TinyUSB CDC

mjan-jansson, 2022-07-17:
TLDR; Does maxed out read/writes cause USB Serial connections to hang?

I am having some problems with a project based on RP2040, using Pico SDK 1.3.1 and latest version of TinyUSB (master branch).  I'm finding somewhat similar posting on other forums, but nothing definitive that would explain what I am seeing. I would thus appreciate comments/help from anyone with insight into Pico SDK and TinyUSB. 

I have narrowed down my issue to a small example, and I am able to reproduce the issue on a Pico board consistently. Given the trivial nature of my example, I would expect this to be a Pico SDK specific issue  (a problem in the raspberry specific code of TinyUSB). 

Before going any further, I would be ever so grateful if someone could comment on and/or try out the code below to confirm that this is reproducible elsewhere. I hate to find out that I have a dodgy USB cable, or a broken RPI4 board. I do not see any electrical issues with USB signals, so I am guessing that this is a software issue (me using the SDK wrong, and a glitch somewhere in the SDK). 

This is the C++ code.  I am  running this from a **debugger** (gdb 11.1), connected via **openocd** (0.11.0-g228ede4-dirty) on a **raspberry pi 4**  (Linux dev1 5.10.103-v7l+), connected to the **Pico** board.:
```
uint8_t usbbuf[512];
int main() {
    tusb_init();
    while (true) {
        tud_task();
        if (tud_cdc_n_available(0)) {
            auto cbRead = tud_cdc_n_read(0, usbbuf, sizeof(usbbuf));
            uint8_t *pb = usbbuf;
            for (int i=0; i<20; i++) {
                auto cb = cbRead;
                while (cb) {
                    auto out = tud_cdc_n_write(0, pb, cb);
                    cb -= out;
                    pb += out;
                    tud_task();
		    // REVIEW: Works if following lines is commented out
                    //tud_cdc_n_read(0, usbbuf, sizeof(usbbuf));
                }
            }
            tud_cdc_n_write_flush(0);
        }
    }
}
```

The code is essentially repeating and replicating any input on the USB serial port to the USB serial output. There is 20 times more data being written than read.

To test the code, I run this on a RPI4:
```
dd if=/dev/random of=/dev/ttyACM0 status=progress
```

Pretty much immediately, the code is stuck in the inner loop because tud_cdc_n_write returns zero, which in turned is caused by the USB endpoint for the CDC is both "locked" and "busy". Looks like a race condition of of competing "read" and "write" operations. 

An observation is that if I add a tud_cdc_n_read within the loop (i.e. balance the ratio of read and write calls), then I do not get the hang. 

I'd be most grateful for any feedback.
Regards
Michael

mjan-jansson, 2022-07-18:
I've tested this with SDK 1.4.0 as well, with the same result.

sdbbs, 2022-07-19:
Hi @mjan-jansson ,

I don't know much about this level of USB, but gave your program a spin.

I had to add this to `CMakeLists.txt`:

    target_include_directories(my_project PRIVATE ${PICO_SDK_PATH}/lib/tinyusb/src)

... and headers, so this is my full main.cpp:

```cpp
#include "pico/stdlib.h"  // set_sys_clock_pll
#include "tusb.h"

uint8_t usbbuf[512];
int main() {

    set_sys_clock_pll(1596000000, 6, 2); // change sys clk to 133 MHz

    //stdio_init_all(); // USB serial init
    tusb_init();
    while (true) {
        tud_task();
        if (tud_cdc_n_available(0)) {
            auto cbRead = tud_cdc_n_read(0, usbbuf, sizeof(usbbuf));
            for (int i=0; i<20; i++) {
                uint8_t *pb = usbbuf;
                auto cb = cbRead;
                while (cb) {
                    auto out = tud_cdc_n_write(0, pb, cb);
                    cb -= out;
                    pb += out;
                    tud_task();
                    // REVIEW: Works if following lines is not commented out
                    //tud_cdc_n_read(0, usbbuf, sizeof(usbbuf));
                }
            }
            tud_cdc_n_write_flush(0);
        }
    }
}
```

I build vs pico-sdk commit 2062372  (SDK 1.3.0); and I cannot see any difference for commented line vs uncommented line - both cases seem to work.

I work in MINGW64 in Windows 10, so when this program runs on the RP2040 Pico, it appears as a plain old USB serial port, for me COM7 (which is `/dev/ttyS6` in MINGW64 bash shell)

So I'd just open `gdb`, and `r`(un) the Pico program from there - then if there is an assertion/hardfault, gdb usually breaks.

Then, did the `dd` in my MINGW64 bash shell - and it kept on going in both cases:

```none
$ dd if=/dev/random of=/dev/ttyS6 status=progress
2423296 bytes (2.4 MB, 2.3 MiB) copied, 4 s, 606 kB/s
```

So, FWIW, I cannot really reproduce "code is stuck in the inner loop".

mjan-jansson, 2022-07-20:
Many thanks @sdbbs for trying it out!

With a PC, I am not able to reproduce the issue either. I'm not too surprised though, given that USB is a host-driven protocol (all interaction between the host and the device is initiated by the host, including things like bandwidth allocations, etc). Using a different host may thus result in that this problem is not seen. The cause could even be a glitch in my RPI4, and be unrelated to RP2040 altogether.

Another note on the scenario: If I stream zeros instead of random data, I am also no longer able to reproduce the issue. One thing that is different between using /dev/zero and /dev/random would be that /dev/random is much slower, which suggest that there is a timing component involved in this issue. 

I have a golden opportunity to debug this issue (since I can repro 100%), so I will do so and post a comment here once I have determined the cause.

kilograham, 2022-07-20:
> Given the trivial nature of my example, I would expect this to be a Pico SDK specific issue (a problem in the raspberry specific code of TinyUSB).

Well the Raspberry Pi specific TinyUSB code is still part of TInyUSB, not the SDK, so please open an issue there

closed 2022-07-20 by kilograham